realsenseai/librealsense #6066: Null frames returned after few hours of running

tispratik, 2020-03-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.12.03.00| 
| Operating System & Version |   Ubuntu 16.04 | 
| Kernel Version (Linux Only)    |  4.15.0-88-generic                                         | 
| Platform                                 |    - |
| SDK Version                            |  -                          | 
| Language                            |  python                         | 
| Segment			|  Robot                          | 

### Issue Description
We're seeing random color/depth frames being returned as None after the camera is being run for few hours. In what cases can null frames be returned?

MartyG-RealSense, 2020-03-16:
RealSense 400 Series camera hardware can run indefinitely so long as they remain within recommended operating temperature tolerances.  The link below discusses temperature tolerances of the cameras and effects of high temperature.

https://support.intelrealsense.com/hc/en-us/community/posts/360037765294/comments/360009701234

If you would like to incorporate a temperature check into Python, the link below offers some guidance.

https://github.com/IntelRealSense/librealsense/issues/866

Temperature is not the only factor in long-run sessions that can affect performance though.  Assuming that the camera remains within operating tolerances, if problems occur then they are more likely to be due to non-camera factors such as USB port / cable, computer hardware or operating system glitches.

tispratik, 2020-03-16:
The camera is placed indoors with ambient temperature. Is there a debug script or some debug information I can read from the camera when the failure happens? This is happening far too often for us now. Almost twice a day.

MartyG-RealSense, 2020-03-16:
Aside from the Python code I linked to earlier, the Librealsense ROS wrapper added the ability to publish temperature monitoring in August 2019 if you are able to make use of ROS.

https://github.com/IntelRealSense/librealsense/issues/4053

https://github.com/IntelRealSense/realsense-ros/pull/878

I'm off-shift for the day now as it is midnight in my time zone but will be happy to continue the conversation when I am back in 7 hours from the time of writing this if you still need support.   Good luck!

tispratik, 2020-03-23:
We are not using ROS and we simply compiled librealsense from stable source.

MartyG-RealSense, 2020-03-23:
Do the errors disappear if the camera hardware is reset, please?

tispratik, 2020-03-24:
rs.version gives 2.33.1. The issue got resolved when we compiled from source.

MartyG-RealSense, 2020-03-24:
Can you confirm please that the problem is now solved?  Thanks!

tispratik, 2020-03-24:
Yes!

closed 2020-03-24 by tispratik

MartyG-RealSense, 2020-03-24:
Great news - thanks so much.  :)