realsenseai/librealsense #6067: D435 depth image out of sync with T265 odom

dschnabel, 2020-03-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 & T265 | 
| Firmware Version                   | D435: 05.12.01.00, T265: 0.2.0.879 | 
| Operating System & Version |   Linux (Raspbian Buster)  | 
| Kernel Version (Linux Only)    |  4.19.97-v7l+  | 
| Platform                                 | Raspberry Pi 4  |
| SDK Version                            |  LibRealSense v2.31.0, RealSense ROS v2.2.12  | 
| Segment			|  Robot  | 

### Issue Description
Hello, I have successfully installed my D435 and T265 camera on my Raspberry Pi 4 which controls a wheeled robot. I had to reduce the resolution and FPS to make it run smoother. But there seems to be a problem that the image from the D435 is not in sync with the odometry from the T265.

I put a video and a more detailed description here: https://github.com/IntelRealSense/realsense-ros/issues/1120

When the robot turns it falsely maps old depth images to the new orientation. I believe this is because the two cameras are not in sync? Anything I can do to get the two cameras in sync? Thanks in advance!

(Usually I try not to post the same question in two different repos. But my original post has gone unnoticed and I was told that here is the place to get attention from Intel.)

dorodnic, 2020-03-17:
Hi @dschnabel 
The two device will never be in perfect optical sync (there is no hardware synchronization between them) So you may need to reduce movement speed, especially when turning. 
However, this also may be due to low fps and perhaps high frame loss. Please make sure you build with `-DCMAKE_BUILD_TYPE=Release` to get better performance and perhaps consider reducing spatial resolution instead of FPS.

dschnabel, 2020-03-18:
@dorodnic many thanks for your thoughts. I believe I'm already at the minimum of resolution as you can see from my [launch file](https://gist.github.com/dschnabel/652b3ffdb34212fc0fa8aba119438e22#file-rovy_d435-launch-L10-L13):
```
color width/height: 424/240
depth width/height: 480/270
```

Or is spatial resolution something different?

> However, this also may be due to low fps and perhaps high frame loss

What could cause frame loss? Is that when the fps is too high and the CPU can't keep up?

> Please make sure you build with -DCMAKE_BUILD_TYPE=Release to get better performance

My Realsense ROS nodes are built with the Release flag, but not librealsense. I will build librealsense with the Release flag too and let you know if that increases the performance.

I noticed that the T265 has `enable_sync` turned off while the D435 has it turned on. As a result the T265 uses the time based on the camera (see [here](https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/src/base_realsense_node.cpp#L1448)) and the D435 uses the system time (see [here](https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/src/base_realsense_node.cpp#L1444)). I'm getting slightly better results when I also use the time based on the camera for the D435. This requires hacking the code since D435 will always run in sync mode when `align_depth` or `enable_pointcloud` are set to true (see [here](https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/src/base_realsense_node.cpp#L541-L542)).

I've also experimented with running the D435 node on a [dedicated CPU core](https://unix.stackexchange.com/a/326585) to ensure nothing gets in the way that could delay the D435 frames. I'm not sure if that made a difference, if it did then just minimally.

dschnabel, 2020-03-19:
@dorodnic rebuilding librealsense with `RELEASE` was a great advice! Now I'm getting a rate of 20 fps for the image/depth frames and the ROS node isn't running at 100% anymore. The data from the two cameras seem to be much more in sync now :+1: 

Thanks for pointing me in the right direction. Closing this issue.

p.s. I found that I'm still getting better results using the camera time of the D435 for the frame timestamp (see my note above).

closed 2020-03-19 by dschnabel