realsenseai/librealsense #6068: When using multiple cameras, do we need to create different pipelines and contexts for each camera?

Dontla, 2020-03-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435| 
| Firmware Version                   | 05.11.15.00| 
| Operating System & Version |  Win10  | 
| Platform                                 | PC  |
| SDK Version                            | 2.29.0                     | 
| Language                            |  python3.6.8                           | 
| Segment			|  Robot                       | 

In the process of using multiple cameras for frame transmission, do we need to create a separate rs.pipeline() for each camera, or just create one? What is the role of the rs.pipeline()? I don't understand it well, can you explain it to me?
And I don't understand what the function of the rs.context() also, do we need to create a separate one for each camera? Or just create one at the beginning of the program?
Best regard!

dorodnic, 2020-03-16:
Hi @Dontla 
https://github.com/IntelRealSense/librealsense/tree/master/examples/multicam 
and https://github.com/IntelRealSense/librealsense/blob/master/doc/api_arch.md
are good resources on this.
In general, one context per App, one pipeline per device.
In addition, there are various limitations on multi-camera support on different platforms (Android, Mac OS, Jetson, etc...) and devices (mainly the T265)

RealSenseCustomerSupport, 2020-03-23:
----------------------------------------------

@Dontla Any other questions with the reply from dorodnic? Looking forward to your reply. Thanks!

RealSenseCustomerSupport, 2020-04-01:
----------------------------------------------

@Dontla Could you please update? Thanks!

Dontla, 2020-04-01:
Thank you very much~~I have got it

closed 2020-04-01 by Dontla