realsenseai/librealsense #6071: Where to get the caffe model and caffe prototxt of distance_to_object.ipynb

demoncoderad, 2020-03-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | v2.33.1 | 
| Operating System & Version |   Win 10 | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13) XX                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.33.1                          | 
| Language                            |  python }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I was using the distance_to_object ipynb (https://hub.gke.mybinder.org/user/intelrealsense-librealsense-3gnudi5f/notebooks/notebooks/distance_to_object.ipynb) to test object detection and distance with D435. A problem arose that I don't know where the model and prototxt are used in the example. The one that I use from https://honingds.com/blog/ssd-single-shot-object-detection-mobilenet-opencv/ gives errer of blob.size is greater than 2. Please Help

MartyG-RealSense, 2020-03-16:
The files referenced in the example are available from the link below:

https://github.com/PINTO0309/MobileNet-SSD-RealSense/tree/master/caffemodel/MobileNetSSD

demoncoderad, 2020-03-19:
I used both files mentioned there. when i run it , it gives an error :
Traceback (most recent call last):
  File "pythonfile.py", line 49, in <module>
    net = cv2.dnn.readNetFromCaffe("MobileNetSSD_deploy.prototxt", "MobileNetSSD_deploy.caffemodel")
cv2.error: OpenCV(4.1.2) C:\projects\opencv-python\opencv\modules\dnn\src\caffe\caffe_io.cpp:1132: error: (-2:Unspecified error) FAILED: fs.is_open(). Can't open "MobileNetSSD_deploy.caffemodel" in function 'cv::dnn::ReadProtoFromBinaryFile'

Please Help Me

MartyG-RealSense, 2020-03-19:
If you have only provided the name of the file (e.g **MobileNetSSD_deploy.prototxt**) then you should change directory to the folder that the MobileNetSSD files are in and launch the script from that folder.  Alternatively, you can provide in the script a full path to the folder that the MobileNetSSD files are stored in.

MartyG-RealSense, 2020-03-28:
This case will be closed after the next 7 days  if there are no further responses.  Thanks!

closed 2020-04-09 by MartyG-RealSense

MartyG-RealSense, 2022-09-02:
Hi @mamoonik  Do you have the two files but still experience the error message mentioned in https://github.com/IntelRealSense/librealsense/issues/6071#issuecomment-601144235 please?

mamoonik, 2022-09-02:
Hi Marty yes I have the same issue :?

I tried the solutions described in your comment. 

Redownloaded the protxt file and the caffee model

as well as moved my testscripts in the same folder as the where the model and protxt file were located in.

Still the same error unfortunately :/

**_cv2.error: OpenCV(3.4.18) D:\a\opencv-python\opencv-python\opencv\modules\dnn\src\caffe\caffe_io.cpp:1126: error: (-2:Unspecified error) FAILED: fs.is_open(). Can't open "MobileNetSSD_deploy.prototxt" in function 'cv::dnn::ReadProtoFromTextFile'_**

HERE IS MY .py script:

import cv2                                # state of the art computer vision algorithms library
import numpy as np                        # fundamental package for scientific computing
import matplotlib.pyplot as plt           # 2D plotting library producing publication quality figures
import pyrealsense2 as rs                 # Intel RealSense cross-platform open-source API
print("Environment Ready")

# Setup:
pipe = rs.pipeline()
cfg = rs.config()
#cfg.enable_device_from_file("../object_detection.bag")
profile = pipe.start(cfg)

# Skip 5 first frames to give the Auto-Exposure time to adjust
for x in range(5):
  pipe.wait_for_frames()
  
# Store next frameset for later processing:
frameset = pipe.wait_for_frames()
color_frame = frameset.get_color_frame()
depth_frame = frameset.get_depth_frame()

# Cleanup:
pipe.stop()
print("Frames Captured")
#color = np.asanyarray(color_frame.get_data())
plt.rcParams["axes.grid"] = False
plt.rcParams['figure.figsize'] = [12, 6]
#plt.imshow(color)
#plt.show()

colorizer = rs.colorizer()
colorized_depth = np.asanyarray(colorizer.colorize(depth_frame).get_data())
plt.imshow(colorized_depth)
#plt.show()

# align = rs.align(rs.stream.color)
# frameset = align.process(frameset)

# # Update color and depth frames:
# aligned_depth_frame = frameset.get_depth_frame()
# colorized_depth = np.asanyarray(colorizer.colorize(aligned_depth_frame).get_data())

# # Show the two frames together:
# images = np.hstack((color, colorized_depth))
# plt.imshow(images)

# Standard OpenCV boilerplate for running the net:
height, width = colorized_depth.shape[:2]
expected = 300
aspect = width / height
resized_image = cv2.resize(colorized_depth, (round(expected * aspect), expected))
crop_start = round(expected * (aspect - 1) / 2)
crop_img = resized_image[0:expected, crop_start:crop_start+expected]
configFile = 'MobileNetSSD_deploy.prototxt.txt*9'
modelFile = 'MobileNetSSD_deploy.caffemodel'
net = cv2.dnn.readNetFromCaffe('MobileNetSSD_deploy.prototxt','MobileNetSSD_deploy.caffemodel')
inScaleFactor = 0.007843
meanVal       = 127.53
classNames = ("background", "aeroplane", "bicycle", "bird", "boat",
              "bottle", "bus", "car", "cat", "chair",
              "cow", "diningtable", "dog", "horse",
              "motorbike", "person", "pottedplant",
              "sheep", "sofa", "train", "tvmonitor")

blob = cv2.dnn.blobFromImage(crop_img, inScaleFactor, (expected, expected), meanVal, False)
net.setInput(blob, "data")
detections = net.forward("detection_out")

label = detections[0,0,0,1]
conf  = detections[0,0,0,2]
xmin  = detections[0,0,0,3]
ymin  = detections[0,0,0,4]
xmax  = detections[0,0,0,5]
ymax  = detections[0,0,0,6]

className = classNames[int(label)]

cv2.rectangle(crop_img, (int(xmin * expected), int(ymin * expected)), 
             (int(xmax * expected), int(ymax * expected)), (255, 255, 255), 2)
cv2.putText(crop_img, className, 
            (int(xmin * expected), int(ymin * expected) - 5),
            cv2.FONT_HERSHEY_COMPLEX, 0.5, (255,255,255))

plt.imshow(crop_img)
plt.show()

MartyG-RealSense, 2022-09-02:
There was also a past case of this error message reported at https://github.com/IntelRealSense/librealsense/issues/4712

As far as I am aware, though a successful solution has not been achieved by RealSense Python users who experience this particular error with the **distance_to_object.ipynb** Jupyter Notebook tutorial, unfortunately.

An alternative means of printing the real-world distance in meters from the center of the camera to an observed object would be to use the SDK instruction **get_distance()** like in the pyrealsense2 script at https://github.com/IntelRealSense/librealsense/issues/9939