realsenseai/librealsense #6072: T265 RsDevice in Unity fails to deliver 200 fps pose when fisheye enabled

RendijsSmukulis, 2020-03-16:
-------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Operating System & Version |  Win 10 | 
| Platform                                 | PC  |
| SDK Version                           |  2.33.1 }                          | 
| Language                              |  unity                          | 

### Issue Description

Using the Unity sample package, the SLAM example uses RsDevice to retrieve pose data from the T265. I've verified this getting ~200 updates a second, as per spec. However, enabling the fisheye streams as well drops the update rate to 30fps. I would have expected either 200 fps, where some frames were just pose, and some had both pose and fisheye; or 230fps of pose or fisheye.

### Measuring the framerate

in `RsDevice.cs` I've modified `WaitForFrames()` to read:

```
    private void WaitForFrames()
    {
        var counter = 0;
        Stopwatch sw = null;

        while (!stopEvent.WaitOne(0))
        {
            using (var frames = m_pipeline.WaitForFrames())
            {
                if (sw == null)
                {
                    sw = Stopwatch.StartNew();
                }

                counter++;

                if (counter % 100 == 0)
                {
                    UnityEngine.Debug.Log($"{counter * 1000f / sw.ElapsedMilliseconds} samples per sec");
                }

                RaiseSampleEvent(frames);
            }
        }
    }
```

This will output 200 with only pose, or 30 with pose+fisheye.

dorodnic, 2020-03-17:
Wait for frames will reduce to lowest common FPS. If you want full framerate please try [start with a callback](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/Intel.RealSense/Pipeline/Pipeline.cs#L114)

RendijsSmukulis, 2020-03-26:
Fair enough, using callback fixes the issue. This could be better documented, though.

Thanks!

closed 2020-03-26 by RendijsSmukulis

neilyoung, 2020-03-26:
@RendijsSmukulis  Lately there was an issue with the callback. In the issue description I put a little script. It was clearly to see, that the callback mechanism was slower and left out frames starting with a specific version. Could you try to double check if you are still/also seeing this? 

https://github.com/IntelRealSense/librealsense/issues/5709

Will check that too tonight, just in case you have some time.

RendijsSmukulis, 2020-03-27:
@neilyoung 

I just ran a test similar to your code in Unity.

Setup:

```
this.ActiveProfile = this.pipeline.Start(cfg, this.RaiseSampleEvent);
[..]
private void RaiseSampleEvent(Frame frame)
{
  Debug.Log($"delta: {(DateTime.UtcNow - new DateTime(1970, 1, 1)).TotalMilliseconds - frame.Timestamp:F3}ms");
}
```

this mostly gives me ~4ms, except some hiccups every 6 or 7 frames where I get ~23ms delay:
[image]

So I'm not seeing the same delays as you. This is on Windows 10, Unity, SDK version 2.33.1.

neilyoung, 2020-03-27:
Thanks for testing. Did the same. MacOS catalina, Python 3.7, SDK 2.33.1, development branch. This script (https://github.com/IntelRealSense/librealsense/issues/5709#issuecomment-578290892) - runs up to 1.7 secs delta, but remains stable. Issue not fixed for me