realsenseai/librealsense #6073: Quaternion

nych9811, 2020-03-16:
Hi.
I'm now studying SLAM, and I want to get quaternion. In D435i calibration tutorial, it says like these.

Rotation - rotation from the left infrared camera to IMU, specified as a 3x3 rotation matrix.

Translation - translation from the left IR camera to IMU, specified as a 3x1 vector in millimeters.

And, I got bias and scale alignment matrix. I know what's the meaning. So, how can I get extrinsic parameters "Rotation and Translation"? Please help me!

Thanks.

MartyG-RealSense, 2020-03-16:
You should be able to get the extrinsic parameters of the IMU using the SDK's **rs-enumerate-devices** tool by using the instruction **rs-enumerate-devices-c** to get calibration information from the device.

https://github.com/IntelRealSense/librealsense/tree/master/tools/enumerate-devices

nych9811, 2020-03-16:
Thanks to your help.

nych9811, 2020-03-16:
Then, I have question.

Thanks to your reference, I studied about them. So, if I get translation matrix, is that mean how IMU sensor rotated then before?

MartyG-RealSense, 2020-03-16:
Could you explain your question more please, as I did not understand the meaning.  I apologise.

I hope you find useful a calibration list from a D435i in the link below.  If you go down the list a long way then you will find a list of extrinsics data from a D435i, including translation.

https://github.com/IntelRealSense/librealsense/issues/4050#issuecomment-495111960

nych9811, 2020-03-16:
Dear @MartyG-RealSense 
I'm beginner of image processing and using depth camera, so I am not good at explaining which one I have hard. So, I'll try more about, and then I'll ask you again.
Thanks.

MartyG-RealSense, 2020-03-16:
You are very welcome.  I really do understand.  Good luck with your learning, and please ask questions any time that you need to.  :)

MartyG-RealSense, 2020-03-28:
This case will be closed after the next 7 days  if there are no further responses.  Thanks!

closed 2020-04-09 by MartyG-RealSense