realsenseai/librealsense #6076: T265 and point cloud position in Unity

amir989, 2020-03-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i and T265| 
| Firmware Version                   | latest | 
| Operating System & Version |  Win (10)  |                                        | 
| Platform                                 | PC |
| SDK Version                            |  2   | 
| Language                            |  C#  unity }                          | 
| Segment			| Smartphone/VR/AR/others  | 

### Issue Description
Hi All,
I have both T265 and D435i to recreate my old slam project [that i did on a Tango phone](https://www.youtube.com/watch?v=e2aEfWxjbZ0). 
basically capturing points with their RGB value and saving it on memory in real-time. My problem is that the SDK in unity does NOT provide point cloud positions in Unity's world that is relative to the T265 camera. in a simple word, if i walk, the points wont walk with me!! Both Tango and ZED camera APIs provide the points position in unity's world relative to the tracking pos .Unfortunately it is missing in Intel's SDK for Unity.

My Solution was to make the points child of the tracking camera by trying:
trackingCamera.transform.localToWorldMatrix.MultiplyPoint3x4(rawpoints[i])
or
trackingCamera.transform.TransformPoint(rawpoints[i])

but it didnt work. the result was weird. 

My question is, what is the math equation for transforming the points to unity's world when using the tracking camera(T265)?
or where can I looking for the math equation(im really not good in math though :) )?

I've been stocked with this issue for weeks now and I really need to fix it soon. 
To RealSense team: Please update the Unity wrapper. its been same since almost a year ago. ive already ordered the new LiDAR sensor and im hopping you guys can update the wrapper to fix the bugs and also add some new features(like having the pose in Unity's world when using tracking camera :D ).   

Best,
Amir

ogoshen, 2020-03-18:
Hi Amir,

Parenting the point cloud (RsPointCloudRenderer) to the camera (RsPoseStreamTransformer) in the hierarchy should do the trick.

Regarding the actual transform used to convert T265's coordinate system to Unity look [here](https://github.com/IntelRealSense/librealsense/blob/1d7ef6ac04394d6183178f6e2dd9304d34bb44ff/wrappers/unity/Assets/RealSenseSDK2.0/Scripts/RsPoseStreamTransformer.cs#L81).

Note that `localToWorldMatrix` and `TransformPoint` are the wrong direction, for cameras you want to use `worldToLocalMatrix` or `InverseTrasformPoint`, so when a camera moves right for instance, objects in its view appear to be moving left.
Generally you do not want to transform the points on the CPU each frame unless you really have too.

amir989, 2020-03-19:
Thanks for reply @ogoshen .

you are right. it would work that way. but the problem will be performance and flexibility when i want to save the points or apply different algorithms to them. its not efficient to work with Unity's mesh component for this case. and that's why im working with the raw points. 

Can you think of other ways?

ogoshen, 2020-03-19:
If possible I'd recommend using the GPU, Unity's compute shaders will be a great fit here.
If you must do it on CPU, consider using the new job system + burst compiler.

Is the original issue resolved? You haven't really specified what the `rawpoints` array is... 

You'll might want to check out how get [unsafe direct access](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/Documentation/cookbook.md#unsafe-direct-access), for the vertices of a `Points` frame this would be the `VertexData` property: https://github.com/IntelRealSense/librealsense/blob/1d7ef6ac04394d6183178f6e2dd9304d34bb44ff/wrappers/csharp/Intel.RealSense/Frames/Points.cs#L36

amir989, 2020-03-19:
Im using compute shader for this app for other reasons. base on my tests, using compute shader for transferring points position and taking them back to CPU is slower than doing it completely on CPU. GetData() is making it slow. Job system is still in preview. but i will definitely use it. 

`rawpoints` is `points.CopyVertices(vertices);`  in this [file](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/unity/Assets/RealSenseSDK2.0/Scripts/RsPointCloudRenderer.cs#L161) which i think has same resault as your link. 

I can explain this in better way on a sample project you have. Open "PointCloudDepthAndColor" scene in the [samples folder](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/unity/Assets/RealSenseSDK2.0/Scenes/Samples). add the tracking camera to it. **don't** make pointcloud game object as tracking camera child. instead, change line [159 ](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/unity/Assets/RealSenseSDK2.0/Scripts/RsPointCloudRenderer.cs#L159) to 
 ```
if (points.VertexData != IntPtr.Zero)
                    {
                        points.CopyVertices(vertices);
                        var temp = vertices.Length;
                        var matrix = GameObject.Find("TrackingCamera").transform.localToWorldMatrix;
                        for (int i = 0; i < temp; i++)
                            vertices[i] = matrix.MultiplyPoint3x4(vertices[i]);
                        mesh.vertices = vertices;
                        mesh.UploadMeshData(false);
                    }
```
 
save it and hit play. To my understanding, this should work same as making the point cloud gameobject as child of the tracking camera. but you will see different behavior. its like the origin point is incorrect!

amir989, 2020-03-20:
I just noticed that even the parenting in the Intel's sample is not correct. in the [video](https://www.youtube.com/watch?v=HwZb7qDsKE0), I move my mouse to the corner of the blue mattress the second time im rotating the camera. and you can see the mattress moving in space when i rotate the camera. same thing is happening in [Keijiro ](https://github.com/keijiro/Rsvfx) sample! here is a [video ](https://www.youtube.com/watch?v=w_aGrJz5FA0&feature=youtu.be)for his sample. I think the problem is in camera image rotation. if you [look ](https://www.youtube.com/watch?v=6e3KA4vJ_Fc&feature=youtu.be) at the right corner in my project, you will see that the image is rotating when im rotating the camera. like the blue mattress turning to left. this should NOT happen. if you look at any other platforms, like Tango, ARCore or ARKit. no matter how you rotate the camera, the camera image would'nt rotate. So the mattress should still look up even if im rotating the physical camera!

Let me know what you think!

ogoshen, 2020-03-22:
Your video makes things much clearer! could have been a bit clearer if you'd have selected Unity's camera in the scene view to show it's frustum.

So the transformation is correct, and the point cloud is generally in place.

To me this looks like "just" temporal syncing. 
Depth and tracking camera aren't synced, also the display and tracking aren't synced, see https://github.com/IntelRealSense/librealsense/issues/5103#issuecomment-547364976.
The TL;DR is, turning vsync off can help, pose prediction (probably based on `UnityEngine.FrameTiming`) should be used.

amir989, 2020-03-23:
Thanks for the response. I'm a bit confused. I'm not sure what do you mean by  

> So the transformation is correct, and the point cloud is generally in place. 

Cause base on the videos i sent you, when im rotating the sensors the points aren't where they should be. they shift to left for same amount. every time I rotate the sensors. 

I have 60fps and the jittery from tracking camera is fine for now. jittery is like 0.00x for me which i can take average during frames to make it less (Vsync is off).

English is my second language and sometimes im not clear. so im gonna try to explain my problems in a different way:

1- How can I get points position in Unity's world, without using Unity's mesh component or parenting while im using tracking camera? 

2- a big shift(offset) happens in points position when im rotating the sensors. How can I fix that shift or where the problem is coming from?(this is different than the jittery problem)

Cheers

amir989, 2020-04-16:
I know the corona makes everything slower. but are these problems something that Intel would consider to revise or fix in close future?

ogoshen, 2020-04-19:
Yeah sorry for the late response.

Looking at your video agin, I might suspect the vertical flip here:
https://github.com/IntelRealSense/librealsense/blob/1d7ef6ac04394d6183178f6e2dd9304d34bb44ff/wrappers/unity/Assets/RealSenseSDK2.0/Shaders/PointCloudGeom.shader#L55

This could be made worse by the extrinsics.
In your video, when you rotate the tracking device, you're doing so freely, and not around the center of tracking. So there's clearly a translation, as seen by the yellow trail.

Now the offset (extrinsics) between the D435 and T265 would cause more such translation when rotating. So the transform node for D435's point cloud should match the real physical setup to compensate.

amir989, 2020-04-28:
Thank you for the answer @ogoshen  

> when you rotate the tracking device, you're doing so freely, and not around the center of tracking. So there's clearly a translation, as seen by the yellow trail. 

what you are saying make a lot sense. But, isnt the rotation data coming from the IMU sensor on T265? if yes, then does it really matter how am I rotating it since its note image based rotation tracking? (im not hte expert im just assuming obviously)

To fix the rotation problem, I think the best answer is to use D435i's IMU data for rotation. and use T265 only for position tracking. I couldnt find a way to access D435i's IMU data. is there a way that I can have access to it?

Cheers

ogoshen, 2020-04-30:
It matters because rotating objects around an arbitrary point causes translation.
If the transforms in Unity don't match those of the real devices, you should expect wrong results.

Anyway, you can get a  `MotionFrame` from a `FrameSet` with:
```cs
fs.FirstOrDefault<MotionFrame>(Stream.Gyro, Format.Xyz32f)
```
And can then extract the 3d vector in various ways, see [MotionFrame](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/Intel.RealSense/Frames/MotionFrame.cs)

RealSenseSupport, 2020-09-21:
Thank you for highlighting the issue outlined regarding T265. We have moved our focus to our next generation of products and consequently, we will not be addressing this issue in the T265.

closed 2020-09-28 by RealSenseSupport