realsenseai/librealsense #6077: Lens deformation in RS depth API

Pixelfahnder, 2020-03-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D435i | 
| Firmware Version                   |05.12.02.100| 
| Operating System & Version |16| 
| Kernel Version (Linux Only)    |4.15.0-88-generic| 
| Platform                                 |Intel NUC|
| SDK Version                            |2.33.1| 
| Language                            |C++| 
| Segment			|Image processing| 

### Issue Description
Due to the example in https://github.com/IntelRealSense/librealsense/tree/master/examples/hello-realsense the depth of the image center point can easily be acquired. Of course the center point is not a real problem as the focal point of the lens will affect the image representation very much but what about a x/y coordinate far away from the center? **Will the depth.get_distance() method take a deformation into account or do you have to calculate the lense physics into the acquired value?** If yes how? I'm aware of the fact that you have to take the focal deformation into account if you like to calculate the proper x/y/z coordinated in the real world but is there any magic done for the proper depth in the API?

dorodnic, 2020-03-17:
Hi @Pixelfahnder 
The depth camera ASIC includes rectification hardware responsible for converting distorted raw infrared image and stretching it to appear without distortion. This applies to depth and IR frames. 
RGB can have some minor distortion, but its practically negligible for RGB sensor used in D415 and D435.

Pixelfahnder, 2020-03-17:
@dorodnic So I don't have to take any of my concerns into account? So on the D435i the depth is calculated out of the IR stereo images if I get you correctly?

Thanks for the information.

dorodnic, 2020-03-17:
Yes, you should not be affected by underlying optics in any way

Pixelfahnder, 2020-03-17:
@dorodnic  Thank you for the information and clearification.

Pixelfahnder, 2020-03-17:
@dorodnic I sadly have to reopen this issue as I have one last question regarding this: Playing around in the RealSense Viewer v2.32.1 I noticed that the depth image has a different field of view than the color image which seems to come from different lenses in front of the cameras. In the ROS connection API I noticed that there is a setting to align the color and depth images. Is this also possible in the Viewer and in the librealsense or do I have to code this myself and if yes in which file could I find a starting point? My understanding of this is that it would align the both fields of view. I ask because I need to detect something in the color image and acquire it's depth in space. And if I try to get the pixels on a percentage distance of the image border the different field of view cases a different depth at the edges of the images as it seems to not be the same point in the image.

dorodnic, 2020-03-17:
Yes, please take a look at [align example](https://github.com/IntelRealSense/librealsense/tree/master/examples/align)
It's not exposed in the Viewer, but API supports it (via rs2::align algorithm)

Pixelfahnder, 2020-03-17:
tHANK YOU VERY MUCH @dorodnic .

closed 2020-03-17 by Pixelfahnder