realsenseai/librealsense #6078: Error set_option to save Point cloud

nauti16, 2020-03-17:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | 5.12.3 | 
| Operating System & Version |   Win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  SDK 2.33.1.1366                        | 
| Language                            |  python                          | 
| Segment			|  Robot                          | 

### Issue Description
I want to save point cloud by using python with some options like keep normals, no color and text type etc. So I'm trying to run export_ply_example.py which is provided in this repository like below link. 

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/export_ply_example.py

But it doesn't work with following error:
```
TypeError                                 Traceback (most recent call last)
<ipython-input-87-6e6576afa718> in <module>
     38     # Set options to the desired values
     39     # In this example we'll generate a textual PLY with normals (mesh is already created by default)
---> 40     ply.set_option(rs.save_to_ply.option_ply_binary, False)
     41     ply.set_option(rs.save_to_ply.option_ply_normals, True)
     42 

TypeError: (): incompatible function arguments. The following argument types are supported:
    1. () -> pyrealsense2.pyrealsense2.option

Invoked with: <class 'pyrealsense2.pyrealsense2.save_to_ply'>
```

I think there is some mismatch between python wrapper and example code. I tried to run the code with SDK 2.31.1 version but it's same.
How can I fix it?

RealSenseCustomerSupport, 2020-04-05:
----------------------------------------------

Hi,

This has been fixed in the following pending PR:

https://github.com/IntelRealSense/librealsense/pull/6119

Thank you

RealSenseCustomerSupport, 2020-04-24:
----------------------------------------------

Hi.

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

closed 2020-05-08 by RealSenseCustomerSupport