realsenseai/librealsense #6079: I don't get frames with a depth map when I connect the camera via USB/IP.

CaBuHoB, 2020-03-17:
Hello!
I am trying to connect the depth camera via USB/IP program over USB 3.0.

After connecting via USB/IP, I can output a 1280x720 30fps RGB8 image without any delays or slowdowns. The speed is ~57000 KB / s.
But if I want to get a depth map at any resolution greater than 480x270, the frames are not transmitted and the speed is ~1000 KB/s.

The server was different devices on Ubuntu 18.04 with USB 3.0. Client - Ubuntu 18.04.

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version |    Linux Ubuntu 18.04  | 
| Kernel Version    |  5.3.0 server and client           | 
| Platform                                 | PC  |
| SDK Version                            |  2.32.1                          | 
| Language                            |  Python  | 
| Segment			|  Tracking |

MartyG-RealSense, 2020-03-17:
RealSense cameras can transmit a high volume of data when higher depth resolutions and / or FPS speeds are being used.  If the camera data is being sent between computers in real-time over a network connection instead of through a USB cable physically linking the two computers then I could foresee that the volume of data being transmitted might become overwhelming once the resolution is above a certain low setting.

If this is the case and a high resolution is important to your application, using data compression, a slower FPS speed or some form of buffering on the network (not trying to transmit the full volume of data at the same time) might improve performance.

MartyG-RealSense, 2020-03-28:
This case will be closed after the next 7 days  if there are no further responses.  Thanks!

closed 2020-04-09 by MartyG-RealSense