realsenseai/librealsense #6081: Obtain an inaccurate X coordinate but the Y and Z coordinate are accurate.

TianchaoHuo, 2020-03-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Operating System & Version |   Linux (Ubuntu 16.04 |                                         | 
| Language                            | python                       | 

### Issue Description
Hello, I am using a D435 camera to compute out the coordinates of interested corner points. As you can see in the picture, the red points can be detected by my algorithms and therefore I can compute out the center of the box. However, thruough the measurement and calculation, the value of X-axis is not accurate, which has 10cm error approximately,  but its Y, and Z coordinates error value are within 1cm around. Note that I already put the camera in front of the center of the box and therefore it is supposed that the X coordinate should be approximate to zero ideally.The function I use is "rs2_deproject_pixel_to_point(intrinsic, [x,y], D) " from libaray pyrealsense2.
[image]

Meanwhile, I am aslo confused about where is exactly the center of the camera. Since using above function "rs2_deproject_pixel_to_point(intrinsic, [x,y], D)", I can directly obtain the X,Y,Z value of interested points. But I am strongly wandering what is the reference coordinate of these points. Are these points coordinates relative to the left orignal depth imager  or the rgb sensor?

could you give me some suggestions that I can deal with this problem? Is it related to the camera precision? or camera calibration? or fine-tune the camera configuration, such as exposure degree and so on...? I have been struggle with this problem a while. Hopefully you can give me a hand.

@chachi @matkatz   @dorodnic @ev-mp

Tom127, 2020-03-18:
@dorodnic @ev-mp @RealSense-Customer-Engineering @MartyG-RealSense 
hi, I encountered exactly the same problem, wrong X but correct Y and Z, especially when there is an angle around the Y axis between the camera and object.
I tried to get the extrinsic matrices with
 `depth_to_color_extrin = aligned_depth_frame.profile.get_extrinsics_to(color_frame.profile)`
`color_to_depth_extrin = color_frame.profile.get_extrinsics_to(aligned_depth_frame.profile)`, 
but got both `([1,0,0], [0,1,0], [0,0,1])`
Any suggestions will be appreciated.

dorodnic, 2020-03-18:
Hi @TianchaoHuo @Tom127 
I'd like to help, but cannot do so without additional details about how these measurements have been collected. Are you seeing similar offset using provided `rs-align` example? Is it possible to share some of the relevant code?

RealSenseCustomerSupport, 2020-03-25:
----------------------------------------------

Please let us know if the rs-align example also exhibit the same behavior or need any further assistance. Thank you.

closed 2020-04-02 by RealSenseCustomerSupport