realsenseai/librealsense #6084: Error when using D435i and t265 together

raghav123456, 2020-03-18:
Hello,
I am trying to run the sample example rs-tracking-and-depth
https://github.com/IntelRealSense/librealsense/tree/master/examples/tracking-and-depth 
but i am getting this error - 
RealSense error calling rs2_pipeline_start_with_config(pipe:0x55eaa29dbc00, config:0x55eaa29db000):
    hwmon command 0x10 failed. Error type:  (1).

i dont understand why this error 
Device used : Tracking Camera ( t265 ) and
                        Depth camera ( D435i )
can someone help?

radfordi, 2020-03-18:
For now this construct is broken:
```C++
    for (auto&& dev : ctx.query_devices())
    {
        rs2::pipeline pipe(ctx);
        rs2::config cfg;
        cfg.enable_device(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
        pipe.start(cfg);
        pipelines.emplace_back(pipe);
    }
```
This is means that `rs-multicam` is broken with `T265` as well.   If we could pass the `device` to `enable_device` for make a list of serial numbers and then run the loops on them, it would work.

The fundamental issues is that `librealsense` doesn't support two `T265` `rs2::device`s at the same time.

RealSenseSupport, 2020-09-21:
Thank you for highlighting the issue with multi-cam using T265. We have moved our focus to our next generation of products and consequently, we will not be addressing this issue any further in the T265.

closed 2020-11-12 by RealSenseSupport