realsenseai/librealsense #6085: build from source

janewoo0123, 2020-03-18:
I want to compile the pyrealsense from source on win10, so I follow the https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python
, but I am confusing that where is the directory I should appoint.

MartyG-RealSense, 2020-03-18:
Detailed guidance for compiling Pyrealsense2 from source on Windows are provided at the link below, which i hope will be helpful to you.

https://github.com/IntelRealSense/librealsense/issues/980

Within that discussion, a RealSense user contributed their own checklist of steps for compiling:

https://github.com/IntelRealSense/librealsense/issues/980#issuecomment-356097825

janewoo0123, 2020-03-19:
my version of cmake is camake-3.17.0-rc3-win64-x64, I run the cmake-gui.exe, and select location where the source is, and press the button of Configure, selecting Visual Studio 15 2017 and x64,
but it will display the warning, I have no idea why.
[image]

MartyG-RealSense, 2020-03-19:
The Windows instructions for building from source recommend CMake 3.8 or newer.  So your 3.17 should be fine.  You also have a compatible version of Visual Studio.

I note that in the CMake log at the bottom of the screen on your kindly provided image, it says 

**Visual Studio 15 2017**
**Could not find any instance of Visual Studio.**

Advice in solving this error is in the link below.

https://stackoverflow.com/questions/51668676/cmake-visual-studio-15-2017-could-not-find-any-instance-of-visual-studio

janewoo0123, 2020-03-20:
I follow the link:
https://stackoverflow.com/questions/51668676/cmake-visual-studio-15-2017-could-not-find-any-instance-of-visual-studio
step:
1.Open the Visual Studio
2.Go to Tools->Get Tools and Features
3.In the "Workloads" tab enable "Desktop development with C++"
4.Click Modify at the bottom right
5.reboot
--------------------------------------------------------------------------
then I find that I have two Windows SDK, so I go to control panel->programs and features to uninstall the older version.
I leave the version of Windows Software Development Kit - Windows 10.0.18362.1
--------------------------------------------------------------------------
then I refer these links:
1.https://github.com/IntelRealSense/librealsense/issues/2719
2.https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python
step:
1.run the cmake-gui
2.select BUILD_PYTHON_BINDINGS and cancel BUILD_WITH_TM2 options.
3.open librealsense2.sln, and build the pyrealsense2 project
4.copy the build output (realsense2.dll and pyrealsense2.pyd) next to your script.
--------------------------------------------------------------------------
finally, it can run.

janewoo0123, 2020-03-20:
but unfortunately, as I tried it on the ubuntu16.04 before, the rs-imu-calibration.py can run, but the procedure wouldn't get into the next step, it would stay at the mode of Status.rotate always.
[image]
[image]

MartyG-RealSense, 2020-03-20:
@janewoo0123   Are you holding the camera in your hand when calibrating, please?   If so, you may find it easier to complete calibration if you attach the camera to the tripod that is supplied with it.  This makes it much more stable, so that you can then adjust the camera turn and rotate and take your hand away from the camera.  It should then hopefully be motionless enough to go on to the next calibration stage.

janewoo0123, 2020-03-23:
I use the tripod to let the camera be stable and adjust the rotate by manual, 
but it still couldn't get into the next procedure.
[image]
but it still couldn't get into the next procedure.

MartyG-RealSense, 2020-03-23:
I recalled a previous case in which the calibrator could not progress past the rotate stage and managed to track down the information.  They too were using the tripod.  The advice I gave, which fixed the problem for the user in that case, was this:

********************

When the status.rotate values are being displayed, the calibration guide says that "You have to bring the numbers to [0,0,0] and then you are in the right direction and the script moves on to the next status".

janewoo0123, 2020-03-24:
I put the camera on the table without the tripod, then run the script, the screen shows that :
-----------------------------------------------------
Start interactive mode:
FOUND ACCEL with fps=63
FOUND GYRO with fps=200
-------------------------
*** Press ESC to Quit ***
-------------------------

Align to direction:  [ 0. -1.  0.]   Upright facing out
   Status.rotate:           [-0.0221 -0.0018 -0.0553]:                 [ True  True  True]
----------------------------------------------------
[image]

----------------------------------------------------
 I let the camera be static while running the script 
but it still couldn't move to the next status.
Is that problem that fps of ACCEL and GYRO is too low resulting in error? 
or the numbers of Status.rotate that I adjust are not too precise to calibrate?

MartyG-RealSense, 2020-03-24:
I guess that the information is asking that in this stage, the camera is rotated til the readout is 0, -1, 0 and then the camera is left alone until the program decides to move on to the next stage.

janewoo0123, 2020-03-24:
I let the camera be upright facing out about ten minutes.
but it still couldn't move to the next status. 
[image]

MartyG-RealSense, 2020-03-24:
The **status:rotate** values are not at the **0, -1, 0** values needed to proceed.  Are the numbers changing when you leave the camera alone, please?

janewoo0123, 2020-03-24:
should the Status.rotate values close to [0 0 0 ]?
, the values after the second decimal place keep changing fast when the camera is static.

MartyG-RealSense, 2020-03-24:
You should follow the instructions at each stage stating which values to align to.  So if it asks for the direction to be 0, -1, 0 for example, try to get the changing values to be 0, -1, 0.

Because the IMU does not have internal calibration, its gravity (Y) may not be zero when the camera is static.  So you may have to hold onto the camera, keeping it as still as possible whilst doing small careful motions.  Holding it by the tripod with the legs folded in so it is like a handle may be the best way to do it.

janewoo0123, 2020-03-25:
position#1-upright facing out 
this pose should show the [True True True ] on the screen, meaning that I place the camera with correct the pose, right?
if I am right as I think, the values should be closely the zero. 
I keep the camera's threaded facing down using three-way to hold.
1.place the camera on the flat table.
[image]

2.place the camera with the tripod on the table.
[image]

3.use hand to hold the tripod.
[image]

MartyG-RealSense, 2020-03-25:
The documentation says that once the values match the target (e.g 0, -1, 0) a three second countdown begins on screen and you have to try not to move the camera too much during the countdown.

After three seconds, a line of dots starts appearing.  When there are twenty dots, the process goes to the next stage.  If the camera is moved away from the target values (e.g 0, -1, 0) then the dots stop increasing.  If the camera moves **too** much then the dot counting cancels and it goes back to **rotate** and you have to begin aligning to the target values again.

Do you ever get to the three second countdown appearing, please?

closed 2020-05-30 by MartyG-RealSense