realsenseai/librealsense #6086: Current development branch does not compile any longer on macOS

neilyoung, 2020-03-18:
...or I'm missing a new CMake setting...

For verification: HEAD is f9480b7567e7bfc3e3a12eeedd42e70c09b26553
My CMake setup is an older one:

```
cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false -DCMAKE_BUILD_TYPE=Release -DBUILD_PYTHON_BINDINGS=ON -DPYTHON_EXECUTABLE=/usr/local/bin/python3

```

Compilation fails at around

`[ 42%] Building CXX object wrappers/python/CMakeFiles/pyrealsense2.dir/python.cpp.o
`

like so:

```
In file included from /Users/decades/Documents/librealsense/wrappers/python/python.cpp:4:
/Users/decades/Documents/librealsense/wrappers/python/python.hpp:155:34: error: constexpr function's return type 'std::string' (aka
      'basic_string<char, char_traits<char>, allocator<char> >') is not a literal type
    static constexpr std::string func() { return py::format_descriptor<typename FmtToType<FMT>::type>::format(); }
                                 ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/string:4322:64: note: 'basic_string<char,
      std::__1::char_traits<char>, std::__1::allocator<char> >' is not literal because it is not an aggregate and has no constexpr constructors other
      than copy or move constructors
_LIBCPP_EXTERN_TEMPLATE(class _LIBCPP_EXTERN_TEMPLATE_TYPE_VIS basic_string<char>)
                                                               ^
9 warnings and 1 error generated.
make[2]: *** [wrappers/python/CMakeFiles/pyrealsense2.dir/python.cpp.o] Error 1
make[1]: *** [wrappers/python/CMakeFiles/pyrealsense2.dir/all] Error 2
make: *** [all] Error 2

```

lramati, 2020-03-18:
it seems that the stdlib/compiler you are using doesn't allow constexpr std::strings. If you remove the constexpr tag on the erroring line, compilation should succeed. I'll make this change in the development branch soon

lramati, 2020-03-18:
What compiler and version are you using?

neilyoung, 2020-03-18:
```
~/Documents/librealsense/build $ /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ --version
Apple clang version 11.0.0 (clang-1100.0.33.17)
Target: x86_64-apple-darwin19.3.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin

```

Will check your suggestion.

neilyoung, 2020-03-18:
Confirmed. Changing just line 155 did do it.

radfordi, 2020-03-18:
@neilyoung, I made a fix (778e89e) for this issue as part of #6083.

neilyoung, 2020-03-18:
@radfordi Thanks, could you please elaborate on #6083? I never noticed it. For a couple of weeks I only used the main release versions. Today I switched to development again, but I didn't notice the 100% load. What I have seen is this: https://github.com/IntelRealSense/librealsense/issues/5709. Is it related to the fix?

radfordi, 2020-03-18:
@neilyoung, you should see the 100% CPU issue even with the release version.  Just try the example program I included in #6083.

neilyoung, 2020-03-18:
Sorry for the dumb question, but where is the test program?

radfordi, 2020-03-18:
@neilyoung, oops, I linked to the PR #6083 instead of the issue #5783 where the test program is.

neilyoung, 2020-03-18:
Hmm... Can't confirm that 100% load with the example app. Is there a special CMake configuration necessary?

Here is mine:

```
cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false -DCMAKE_BUILD_TYPE=Release -DBUILD_PYTHON_BINDINGS=ON -DPYTHON_EXECUTABLE=/usr/local/bin/python3
```

radfordi, 2020-03-18:
No @neilyoung, nothing special required.

neilyoung, 2020-03-19:
Hmm. Then I can't confirm the 100 % load. And the #6083 is not part of the branch I'm using currently (master)

neilyoung, 2020-03-20:
The initial problem of this thread is solved by the header patch and/or #6083

closed 2020-03-20 by neilyoung