realsenseai/librealsense #6087: Crash android App when use Wrappers Android SDK

shalkov, 2020-03-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D434i  | 
| Firmware Version                   | 05.12.01.00 | 
| Operating System & Version |   Android 9  | 
| Platform                                 | Android Phone |
| SDK Version                            |  2.33.1                          | 
| Language                            | kotlin                        | 
| Segment			| Smartphone                       | 

### Issue Description
Please help!
When i start my android app and use RealSense Camera together.
At some point, I get the following error and the application crashes:
```
    --------- beginning of crash
A/libc: Fatal signal 11 (SIGSEGV), code 1, fault addr 0x0 in tid 6059 (tudio.mytestapp), pid 5794 (tudio.mytestapp)
I/crash_dump32: obtaining output fd from tombstoned, type: kDebuggerdTombstone
I//system/bin/tombstoned: received crash request for pid 5794
I/crash_dump32: performing dump of process 5794 (target tid = 6059)
A/DEBUG: *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
A/DEBUG: Build fingerprint: 'TES/T2234AS/T2234AS:8.1.0/OPM6.171019.030.E1/151221:userdebug/test-keys'
A/DEBUG: Revision: '0'
A/DEBUG: ABI: 'arm'
A/DEBUG: pid: 5794, tid: 6059, name: tudio.mytestapp  >>> ru.studio.mytestapp <<<
A/DEBUG: signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0
A/DEBUG: Cause: null pointer dereference
A/DEBUG:     r0 87e7dcc0  r1 00000000  r2 895bc6c5  r3 00000000
A/DEBUG:     r4 00000002  r5 87e7dba0  r6 87e7dba0  r7 6519aca0
A/DEBUG:     r8 00000000  r9 83143898  sl 6519aaf4  fp a85201b8
A/DEBUG:     ip 897c536c  sp 6519a990  lr 8957028d  pc 8957f0c2  cpsr 60070030
A/DEBUG: backtrace:
A/DEBUG:     #00 pc 004370c2  /data/app/ru.studio.mytestapp-jObrMy43AUlNq8UnjLX93g==/lib/arm/librealsense2.so (librealsense::synthetic_source::allocate_composite_frame(std::__ndk1::vector<librealsense::frame_holder, std::__ndk1::allocator<librealsense::frame_holder>>)+273)
A/DEBUG:     #01 pc 0045a0e3  /data/app/ru.studio.mytestapp-jObrMy43AUlNq8UnjLX93g==/lib/arm/librealsense2.so (librealsense::pipeline::aggregator::handle_frame(librealsense::frame_holder, librealsense::synthetic_source_interface*)+706)
A/DEBUG:     #02 pc 0045a4f1  /data/app/ru.studio.mytestapp-jObrMy43AUlNq8UnjLX93g==/lib/arm/librealsense2.so
A/DEBUG:     #03 pc 004359d7  /data/app/ru.studio.mytestapp-jObrMy43AUlNq8UnjLX93g==/lib/arm/librealsense2.so (librealsense::processing_block::invoke(librealsense::frame_holder)+54)
A/DEBUG:     #04 pc 00454295  /data/app/ru.studio.mytestapp-jObrMy43AUlNq8UnjLX93g==/lib/arm/librealsense2.so
A/DEBUG:     #05 pc 004bdd5b  /data/app/ru.studio.mytestapp-jObrMy43AUlNq8UnjLX93g==/lib/arm/librealsense2.so (librealsense::frame_source::invoke_callback(librealsense::frame_holder) const+98)
A/DEBUG:     #06 pc 004367e3  /data/app/ru.studio.mytestapp-jObrMy43AUlNq8UnjLX93g==/lib/arm/librealsense2.so (librealsense::synthetic_source::frame_ready(librealsense::frame_holder)+26)
A/DEBUG:     #07 pc 00439fb1  /data/app/ru.studio.mytestapp-jObrMy43AUlNq8UnjLX93g==/lib/arm/librealsense2.so
A/DEBUG:     #08 pc 004359d7  /data/app/ru.studio.mytestapp-jObrMy43AUlNq8UnjLX93g==/lib/arm/librealsense2.so (librealsense::processing_block::invoke(librealsense::frame_holder)+54)
A/DEBUG:     #09 pc 00454373  /data/app/ru.studio.mytestapp-jObrMy43AUlNq8UnjLX93g==/lib/arm/librealsense2.so
A/DEBUG:     #10 pc 004b9a7d  /data/app/ru.studio.mytestapp-jObrMy43AUlNq8UnjLX93g==/lib/arm/librealsense2.so
A/DEBUG:     #11 pc 004bdd5b  /data/app/ru.studio.mytestapp-jObrMy43AUlNq8UnjLX93g==/lib/arm/librealsense2.so (librealsense::frame_source::invoke_callback(librealsense::frame_holder) const+98)
A/DEBUG:     #12 pc 004367e3  /data/app/ru.studio.mytestapp-jObrMy43AUlNq8UnjLX93g==/lib/arm/librealsense2.so (librealsense::synthetic_source::frame_ready(librealsense::frame_holder)+26)
A/DEBUG:     #13 pc 004993c7  /data/app/ru.studio.mytestapp-jObrMy43AUlNq8UnjLX93g==/lib/arm/librealsense2.so (rs2_synthetic_frame_ready+50)
A/DEBUG:     #14 pc 00437cfd  /data/app/ru.studio.mytestapp-jObrMy43AUlNq8UnjLX93g==/lib/arm/librealsense2.so (rs2::frame_source::frame_ready(rs2::frame) const+32)
A/DEBUG:     #15 pc 00437be1  /data/app/ru.studio.mytestapp-jObrMy43AUlNq8UnjLX93g==/lib/arm/librealsense2.so
A/DEBUG:     #16 pc 004359d7  /data/app/ru.studio.mytestapp-jObrMy43AUlNq8UnjLX93g==/lib/arm/librealsense2.so (librealsense::processing_block::invoke(librealsense::frame_holder)+54)
A/DEBUG:     #17 pc 004b9dc9  /data/app/ru.studio.mytestapp-jObrMy43AUlNq8UnjLX93g==/lib/arm/librealsense2.so
A/DEBUG:     #18 pc 004bdd5b  /data/app/ru.studio.mytestapp-jObrMy43AUlNq8UnjLX93g==/lib/arm/librealsense2.so (librealsense::frame_source::invoke_callback(librealsense::frame_holder) const+98)
A/DEBUG:     #19 pc 004b4695  /data/app/ru.studio.mytestapp-jObrMy43AUlNq8UnjLX93g==/lib/arm/librealsense2.so
A/DEBUG:     #20 pc 00427c27  /data/app/ru.studio.mytestapp-jObrMy43AUlNq8UnjLX93g==/lib/arm/librealsense2.so (std::__ndk1::function<void (librealsense::platform::stream_profile, librealsense::platform::frame_object, std::__ndk1::function<void ()>)>::operator()(librealsense::platform::stream_profile, librealsense::platform::frame_object, std::__ndk1::function<void (librealsense::platform::stream_profile, librealsense::platform::frame_object, std::__ndk1::function<void ()>)>) const+70)
A/DEBUG:     #21 pc 0046eb11  /data/app/ru.studio.mytestapp-jObrMy43AUlNq8UnjLX93g==/lib/arm/librealsense2.so
A/DEBUG:     #22 pc 003db363  /data/app/ru.studio.mytestapp-jObrMy43AUlNq8UnjLX93g==/lib/arm/librealsense2.so (std::__ndk1::function<void (dispatcher::cancellable_timer)>::operator()(dispatcher::cancellable_timer) const+18)
A/DEBUG:     #23 pc 00467097  /data/app/ru.studio.mytestapp-jObrMy43AUlNq8UnjLX93g==/lib/arm/librealsense2.so (_ZZN13active_objectINSt6__ndk18functionIFvN10dispatcher17cancellable_timerEEEEE7do_loopEvENKUlS3_E_clES3_+8)
A/DEBUG:     #24 pc 003db363  /data/app/ru.studio.mytestapp-jObrMy43AUlNq8UnjLX93g==/lib/arm/librealsense2.so (std::__ndk1::function<void (dispatcher::cancellable_timer)>::operator()(dispatcher::cancellable_timer) const+18)
A/DEBUG:     #25 pc 003db229  /data/app/ru.studio.mytestapp-jObrMy43AUlNq8UnjLX93g==/lib/arm/librealsense2.so (_ZZN10dispatcherC1EjENKUlvE_clEv+72)
A/DEBUG:     #26 pc 003db189  /data/app/ru.studio.mytestapp-jObrMy43AUlNq8UnjLX93g==/lib/arm/librealsense2.so (_ZNSt6__ndk114__thread_proxyINS_5tupleIJNS_10unique_ptrINS_15__thread_structENS_14default_deleteIS3_EEEEZN10dispatcherC1EjEUlvE_EEEEEPvSA_+36)
A/DEBUG:     #27 pc 00047d3f  /system/lib/libc.so (__pthread_start(void*)+22)
A/DEBUG:     #28 pc 0001b029  /system/lib/libc.so (__start_thread+32)
E//system/bin/tombstoned: Tombstone written to: /data/tombstones/tombstone_09
```
I don’t know the exact playback steps.
the error appears it is not clear when, but exactly when I display the image on the screen

MartyG-RealSense, 2020-03-20:
This is the first error in the backtrace list:

A/DEBUG:     #00 pc 004370c2  /data/app/ru.studio.mytestapp-jObrMy43AUlNq8UnjLX93g==/lib/arm/librealsense2.so (librealsense::synthetic_source::allocate_composite_frame(std::__ndk1::vector<librealsense::frame_holder, std::__ndk1::allocator<librealsense::frame_holder>>)+273)

There has been another case of a Librealsense Android user who had a frame_holder error in December 2019.  Similarly to yourself, their application would start running and then crash later.  The amount of time that it ran before crashing seemed to be related to the resolution being used.  The higher the resolution, the sooner it crashed.

https://github.com/IntelRealSense/librealsense/issues/5524#issuecomment-570512290

This behaviour is suggestive of a memory leak.  The RealSense SDK currently has some fixes for memory leaks on its 'pull' list for introduction in an upcoming version of the SDK.

https://github.com/IntelRealSense/librealsense/pull/5910

memory leak

MartyG-RealSense, 2020-05-26:
Case closed due to reported bug not being reported by other users recently.

closed 2020-05-26 by MartyG-RealSense