realsenseai/librealsense #6088: Intel d415 IR stream in RGB coordinates

talbenach, 2020-03-18:
Is it possible to obtain the depth stream from the d415 camera converted to the RGB camera's coordinates at the source? i.e. similar to what the "align" function does, but without the computing resource investment.
Thanks you

MartyG-RealSense, 2020-03-18:
Not as far as I know.  The Vision Processor D4 hardware component inside the camera performs operations such as image rectification, described in the brochure image in the link below, and then the data travels through the USB cable to the computing device.

https://www.stemmer-imaging.com/media/uploads/cameras/13/136603-Intel-RealSense-D4-Vision-Processor-Brochure.pdf

If freeing up CPU resources is a concern for you, then you could try "offloading" CPU work onto a GPU.   For devices with an Nvidia GPU, this can be done by building Librealsense with CUDA support.  

For non-Nvidia GPUs, there is alternatively "GLSL Processing Blocks" in Librealsense, though this method may be ineffective on low-power devices.

https://github.com/IntelRealSense/librealsense/issues/4905#issuecomment-533854888

MartyG-RealSense, 2020-03-28:
This case will be closed after the next 7 days  if there are no further responses.  Thanks!

closed 2020-04-09 by MartyG-RealSense