realsenseai/librealsense #6089: Android Colorized Depth to bitmap

Murded, 2020-03-18:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415  | 
| Operating System & Version |   Android 10 | 
| Platform                                 | Mobile Phone  |
| Language                            |  Java  | 

### Issue Description
Hi I'm trying to convert the colourised depth image into an android bitmap but I'm having some issues, I believe because of the image format.

I have the following code:
```
public static Bitmap VideoFrameToBitmap(VideoFrame frame)
{
        int size = frame.getStride() * frame.getHeight();
        ByteBuffer mBuffer = ByteBuffer.allocate(size);
        frame.getData(mBuffer.array());
        Bitmap b = Bitmap.createBitmap(frame.getWidth(),frame.getHeight(), Bitmap.Config.ARGB_8888);
        b.copyPixelsFromBuffer(mBuffer);
        return b;
}
```

Which works fine for my RGBA Colour image, however does not work with the colourised depth image (I'm assuming because the format is RGB instead)

I have tried changing ```Bitmap.Config.ARGB_8888``` to the various different options available however none gave the correct result

The code below is how I'm obtaining the colourised depth image:
```
FrameSet aligned = mPipeline.waitForFrames(1000).releaseWith(frameReleaser)
                                 .applyFilter(colorizer).releaseWith(frameReleaser)
                                 .applyFilter(align).releaseWith(frameReleaser);

VideoFrame colourizedFrame = aligned.first(StreamType.DEPTH, StreamFormat.RGB8).releaseWith(frameReleaser).as(Extension.VIDEO_FRAME);
```

I've also tried changing it to ```aligned.first(StreamType.DEPTH, StreamFormat.RGBA8``` however this returns null

MartyG-RealSense, 2020-03-20:
I hope that the link below will give you helpful guidance.

https://github.com/IntelRealSense/librealsense/issues/5446#issuecomment-565688597

Murded, 2020-03-23:
> I hope that the link below will give you helpful guidance.
> 
> [#5446 (comment)](https://github.com/IntelRealSense/librealsense/issues/5446#issuecomment-565688597)

Hi @MartyG-RealSense , Thanks for your help!

I've made a bit of progress and I'm pretty close to what I'm trying to achieve however i'm having trouble obtaining anything other than a black / white image.

Here's what i have at the moment:

```
//This code doesn't change between examples
FrameSet aligned = mPipeline.waitForFrames(1000).releaseWith(frameReleaser)
                        .applyFilter(decimationFilter).releaseWith(frameReleaser)
                        .applyFilter(depthToDisparity).releaseWith(frameReleaser)
                        .applyFilter(spatialFilter).releaseWith(frameReleaser)
                        .applyFilter(temporalFilter).releaseWith(frameReleaser)
                        .applyFilter(disparityToDepth).releaseWith(frameReleaser);

//Apply the align filter
aligned = aligned.applyFilter(align).releaseWith(frameReleaser);

//Code copied from the link provided
Mat testMap = new Mat(depthFrame.getHeight(), depthFrame.getWidth(),CV_16UC1);
                    int size = ((int)testMap.total() * (int)testMap.elemSize());
                    byte[] return_buff = new byte[size];
                    depthFrame.getData(return_buff);
                    short[] shorts = new short[size / 2];
                   ByteBuffer.wrap(return_buff).order(ByteOrder.LITTLE_ENDIAN).asShortBuffer().get(shorts);
                    testMap.put(0, 0, shorts);

Mat mDepth8UC1 = new Mat(testMap.rows(),testMap.cols(),CV_8UC3);
testMap.convertTo(mDepth8UC1,CV_8UC3,1);

bitmap = Bitmap.createBitmap(mDepth8UC1.cols(), mDepth8UC1.rows(), Bitmap.Config.ARGB_8888);
Utils.matToBitmap(mDepth8UC1, bitmap);

//Update the UI
runOnUiThread(() -> imageView.setImageBitmap(bitmap));
```
This code gives the following result:
[image]

However, If I apply the colorizer (Like so): 
```
aligned = aligned.applyFilter(colorizer).releaseWith(frameReleaser);
```
I get the following result:
[image]

With the colorizer still applied I can change the following code to
```
Mat testMap = new Mat(depthFrame.getHeight(), depthFrame.getWidth(),CV_16UC3);
```
I get the following output which is a bit closer to what I want (as it actually has colour this time)

[image]

I've tried a few different combinations and I can't quite work it out, any help would be very appreciated

MartyG-RealSense, 2020-03-23:
I researched your question carefully.  I found an example of how a colorizer can be applied in an Android application:

https://github.com/IntelRealSense/librealsense/issues/3295

[image]

Murded, 2020-03-23:
Hi @MartyG-RealSense do you know if there is a way to do it without using native code? Currently all of our camera streaming is in java and seems like it could be a bit messy / hack to have the camera frames passed from java to the native code just to colour them.

Thanks

-Andrew

MartyG-RealSense, 2020-03-23:
I am still learning Android, so I apologise.   I did further research and found an example RealSense Android capture program in Java that makes use of the colorizer.

I recommend visiting the script in the link below and doing a find on the browser page for 'colorizer' to highlight the sections of the Java script that use it.

https://github.com/leggedrobotics/librealsense-rsl/blob/master/wrappers/android/examples/capture/src/main/java/com/intel/realsense/capture/MainActivity.java

Murded, 2020-03-23:
Hey @MartyG-RealSense I've looked through the source before to see how the renderer works but it uses opengl If I recall correctly which I'm not familliar with.

However, I did manage to resolve the problem using the code below:

```
//Get the frames
FrameSet aligned = mPipeline.waitForFrames(1000).releaseWith(frameReleaser)
                     .applyFilter(decimationFilter).releaseWith(frameReleaser)
                     .applyFilter(depthToDisparity).releaseWith(frameReleaser)
                     .applyFilter(spatialFilter).releaseWith(frameReleaser)
                     .applyFilter(temporalFilter).releaseWith(frameReleaser)
                     .applyFilter(disparityToDepth).releaseWith(frameReleaser)
                     .applyFilter(align).releaseWith(frameReleaser)
                     .applyFilter(colorizer).releaseWith(frameReleaser);

//Grab the depth frame as a video frame
VideoFrame depthFrame = aligned.first(StreamType.DEPTH).releaseWith(frameReleaser).as(Extension.VIDEO_FRAME);

//Create and insert the data into a mat
Mat testMap = new Mat(depthFrame.getHeight(), depthFrame.getWidth(),CV_8UC3);
int size = ((int)testMap.total() * (int)testMap.elemSize());
 byte[] return_buff = new byte[size];
depthFrame.getData(return_buff);
ByteBuffer.wrap(return_buff).order(ByteOrder.LITTLE_ENDIAN).asReadOnlyBuffer().get(return_buff);
testMap.put(0, 0, return_buff);

//Create a bitmap to hold our result                    
bitmap = Bitmap.createBitmap(testMap.cols() , testMap.rows(), Bitmap.Config.ARGB_8888);

//Convert the mat back to a bitmap
Utils.matToBitmap(testMap, bitmap);

//Update the imageview with our bitmap
runOnUiThread(() -> imageView.setImageBitmap(bitmap));
```

The biggest change I made was using a byte array instead of a short array and changing the mat type to CV_8UC3

[image]

MartyG-RealSense, 2020-03-23:
Awesome news - thanks so much for sharing your method and code with the community!

Murded, 2020-03-23:
> Awesome news - thanks so much for sharing your method and code with the community!

Thank you soo much for your help!

closed 2020-03-23 by Murded

bjain409, 2021-03-15:
Hi @Murded @MartyG-RealSense 
related to -
[this](https://github.com/IntelRealSense/librealsense/issues/8551)
I am need of depth of an object I am not interested in the color of the object 
I mean via real sense application whenever we select a point of depth view screen we get the related depth of that object.
for the depth camera, I am not interested in the bitmap of it.
Whenever I select a pixel from the image I should get the related depth of that pixel best if it returns x, y, z-axis distance.
I want a matrix that has pixels and their depth as metadata in the form x, y, z-axis if possible, or only the actual perpendicular distance from real sense depth sensor to object will be fine.
My requirement is I want a bitmap from the RGB camera and a matrix from the depth sensor whenever I trigger a capture event without UI from the background service.

Please revert asap.
Thanks

bjain409, 2021-03-15:
> Hi @Murded @MartyG-RealSense
> related to -
> [this](https://github.com/IntelRealSense/librealsense/issues/8551)
> While triggering the capture session I am need of depth of an object I am not interested in the color of the object from the depth sensor
> I mean via real sense application whenever we select a point of depth view screen we get the related depth of that object.
> for the depth camera, I am not interested in the bitmap of it.
> Whenever I select a pixel from the image I should get the related depth of that pixel best if it returns x, y, z-axis distance.
> I want a matrix that has pixels and their depth as metadata in the form x, y, z-axis if possible, or only the actual perpendicular distance from real sense depth sensor to object will be fine.
> My requirement is I want a bitmap from the RGB camera and a matrix from the depth sensor whenever I trigger a capture event without UI from the background service.
> 
> Please revert asap.
> Thanks

@MartyG-RealSense @Murded 
I am not very much familiar with android please pardon me for that
One more thing:  as I don't need any UI component and rendering
So I don't think whether there will be a need for a handler. Correct if me I am wrong.
On which thread it should work without any freeze.
And even let me know where should be permission granted for USB 
@Murded @MartyG-RealSense Can you   please create a method that  when called
create a Toast msg which has string encoding of RGB camera and matrix of the depth camera.
I know it won't show the encoding string.
Just print the first few characters.
Let me know by when  I can expect it to be done
Thanks.

MartyG-RealSense, 2021-03-15:
Hi @bjain409   I am not involved in development and so cannot write a programming solution for your Android problem.

bjain409, 2021-03-16:
> Hi @bjain409 I am not involved in development and so cannot write a programming solution for your Android problem.

@MartyG-RealSense How do I get my problem solved then?
I am using your products there should be a solution for it.

MartyG-RealSense, 2021-03-16:
@bjain409   I am not an Android specialist either.  I will seek the advice of someone with greater Android programming knowledge.

@kafan1986  Do you kindly have any advice for @bjain409 regarding their question above " Can you please create a method that when called creates a Toast msg which has string encoding of RGB camera and matrix of the depth camera" - thanks!

kafan1986, 2021-03-17:
@bjain409 Are you planning to do processing on a single pair of depth and RGB image? If yes, then you can build a solution like you suggested as it will not be suitable for a real time processing pipeline. Aligning depth w.r.t. to RGB or vice versa is quite computation intensive process. May I know your use case?

bjain409, 2021-03-17:
> @bjain409 Are you planning to do processing on a single pair of depth and RGB image? If yes, then you can build a solution like you suggested as it will not be suitable for a real time processing [...]

@kafan1986 @MartyG-RealSense 
I am in need of the RGB camera image and depth sensor pixel-depth matrix which is captured simultaneously with two threads working at the same time.
If its computing challenge both running simultaneously then I can adjust with some delay in both capture 
If needed I can wait for 500 milliseconds between two capture(Not more than that).
I am ok if fps is reduced. Don't know whats fps for RGB and depth sensors for now.
My actual use case is. I am working on face detection. No doubt the RGB camera is enough for that. But it won't guarantee whether the face detected is real or just the image(2D). (Printed photo) 
Backend library will detect face from RGB sensor create bounding box against that face.
With those corresponding pixels(x, y axis) I need a depth sensor capture which will return depth from sensor to pixel.
For instance, take an example of a human nose tip. Consider the tip of the nose has location (2, 36). So whenever I pass the pixel location (2,36) to the depth sensor matrix I should get the distance of that pixel from the depth sensor.
In basic, I just want depth sensor matrix and RGB image whenever I trigger a capture from the background.
And NO UI!!
Even please consider the permission where it should be placed.

Thanks.

kafan1986, 2021-03-17:
> @kafan1986 @MartyG-RealSense
> I am in need of the RGB camera image and depth sensor pixel-depth matrix which is captured simultaneously with two threads working at the same time.
> If its computing [...]

I understand your requirements. For your use case don't align the entire depth and rgb data. I believe you will need depth information for limited number of pixels for each face detected (corner of face rectangle and may be few 10s of facial landmarks) from RGB image. This is very less compared to aligning entire image, which actually performs this alignment for entire w x h pixels.

Let me suggest you the correct approach. You will have one thread (thread 1) running that captures pair of frame data (RGB + depth) and now you need to copy this frame data to another format (own datastructure), so that you can manage and release the memory as desired, given the intel SDK will release the frame data. Again copying frame data into own datastructure can be little slow, compared to some OpenCV's Mat, so I would prefer it.

Now you can work on this data from another thread (thread 2). You can achieve this using just two sets of (own datastructure) and use a release and lock mechanism. So that either of the datastructure is available readily for copying the latest captured pair of frame to (thread 1), and and the other one can be locked for further image processing. (for idea see how opencv creates a pipeline for image processing from camera on android).

If you still face problem, let me know, I will share some relevant code, although I would have to search for it, when I return back home.

bjain409, 2021-03-17:
> I understand your requirements. For your use case don't align the entire depth and rgb data. I believe you will need depth information for limited number of pixels for each face detected (corner of [...]

@kafan1986 
Thanks for your response.

Face detection is just part of the project and not part of the android app.
I just wanted to share a use case that would have been helpful for you to understand.
Face detection won't happen at the android end. So please don't consider here.
Just help me with capturing data and I am good. 
After capturing I will store the image and depth matrix locally on android SQLite.
Please get me a method that gives me these two objects at the same time if possible(simultaneous).
Or I can manage if the time difference between two capture is less than 100milliseconds.

bjain409, 2021-03-19:
> @kafan1986
> Thanks for your response.
>
> Face detection is just part of the project and not part of the android app.
> I just wanted to share a use case that would have been helpful for you to [...]

@kafan1986 @Murded @MartyG-RealSense 
Please respond.

kafan1986, 2021-03-21:
> @kafan1986 @Murded @MartyG-RealSense
> Please respond.
[Streamer.java.zip](https://github.com/IntelRealSense/librealsense/files/6178373/Streamer.java.zip)

Murded, 2021-03-21:
@kafan1986 I haven't read through all of your requirements but this should be more than enough to set you on the right track.

How to get the depth buffer:
```
//Get the frames from the camera
FrameSet aligned  = mPipeline.waitForFrames(5000).releaseWith(frameReleaser).applyFilter(align).releaseWith(frameReleaser);

 VideoFrame depthFrame = aligned.first(StreamType.DEPTH).releaseWith(frameReleaser).as(Extension.VIDEO_FRAME);

//Note: I Use a Mat here to calculate the size but you can easily calculate this manually but I don't have time to test this
Mat testMap = new Mat(depthFrame.getHeight(), depthFrame.getWidth(),CV_8UC3);
int size = ((int)testMap.total() * (int)testMap.elemSize());

//Create a new byte array to hold the depth data
byte[] return_buff = new byte[size];

//Populate the array with the data
depthFrame.getData(return_buff);

//Note: Now that you have the depth data, you can use intrinsic information to calculate the distance at a given point. I have not tested getting distance on the Java data, you may find that this will not work as expected because you can't have unsigned values in java and may need to write a JNI method to do what you need. 
```
How to get colour bitmap:
```
//Get the frames from the camera
FrameSet aligned  = mPipeline.waitForFrames(5000).releaseWith(frameReleaser).applyFilter(align).releaseWith(frameReleaser);

VideoFrame vf = aligned.first(StreamType.COLOR).releaseWith(frameReleaser).as(Extension.VIDEO_FRAME);

//Convert the releasense video frame to a bitmap we can display

 int size = frame.getStride() * frame.getHeight();
        ByteBuffer mBuffer = ByteBuffer.allocate(size);
        frame.getData(mBuffer.array());
        Bitmap b = Bitmap.createBitmap(frame.getWidth(),frame.getHeight(), Bitmap.Config.ARGB_8888);
        b.copyPixelsFromBuffer(mBuffer);`
``

bjain409, 2021-03-23:
> @kafan1986 I haven't read through all of your requirements but this should be more than enough to set you on the right track.
> 
> How to get the depth buffer:
> 
> ```
> //Get the frames from the camera
> FrameSet aligned  = mPipeline.waitForFrames(5000).releaseWith(frameReleaser).applyFilter(align).releaseWith(frameReleaser);
> 
>  VideoFrame depthFrame = aligned.first(StreamType.DEPTH).releaseWith(frameReleaser).as(Extension.VIDEO_FRAME);
> 
> //Note: I Use a Mat here to calculate the size but you can easily calculate this manually but I don't have time to test this
> Mat testMap = new Mat(depthFrame.getHeight(), depthFrame.getWidth(),CV_8UC3);
> int size = ((int)testMap.total() * (int)testMap.elemSize());
> 
> //Create a new byte array to hold the depth data
> byte[] return_buff = new byte[size];
> 
> //Populate the array with the data
> depthFrame.getData(return_buff);
> 
> //Note: Now that you have the depth data, you can use intrinsic information to calculate the distance at a given point. I have not tested getting distance on the Java data, you may find that this will not work as expected because you can't have unsigned values in java and may need to write a JNI method to do what you need. 
> ```
> 
> How to get colour bitmap:
> 
> ```
> //Get the frames from the camera
> FrameSet aligned  = mPipeline.waitForFrames(5000).releaseWith(frameReleaser).applyFilter(align).releaseWith(frameReleaser);
> 
> VideoFrame frame = aligned.first(StreamType.COLOR).releaseWith(frameReleaser).as(Extension.VIDEO_FRAME);
> 
> //Convert the releasense video frame to a bitmap we can display
> 
>  int size = frame.getStride() * frame.getHeight();
>         ByteBuffer mBuffer = ByteBuffer.allocate(size);
>         frame.getData(mBuffer.array());
>         Bitmap b = Bitmap.createBitmap(frame.getWidth(),frame.getHeight(), Bitmap.Config.ARGB_8888);
>         b.copyPixelsFromBuffer(mBuffer);`
> ``
> ```

@kafan1986 
Thank you for responding back.
I didn't completely follow this:

### 
**__"
//Note: I Use a Mat here to calculate the size but you can easily calculate this manually but I don't have time to test this
Mat testMap = new Mat(depthFrame.getHeight(), depthFrame.getWidth(),CV_8UC3);
int size = ((int)testMap.total() * (int)testMap.elemSize());
"__**

Which is the intrinsic way by which I will get the depth of individual pixel(point)?
Can you please share that?
Should I keep ```

Config cfg = new Config()) {
    cfg.enableStream(StreamType.DEPTH, 640, 480)
   cfg.enableStream(StreamType.COLOR, 640, 480);
    PipelineProfile pp = mPipeline.start(cfg)
}
```
in  onAttach();?

Can you please let me know: the code you shared will return objects of the same time?
I am in need of depth and RGB sensor of same time
Will this work?

`
FrameSet aligned  = mPipeline.waitForFrames(5000).releaseWith(frameReleaser).applyFilter(align).releaseWith(frameReleaser);

 VideoFrame depthFrame = aligned.first(StreamType.DEPTH).releaseWith(frameReleaser).as(Extension.VIDEO_FRAME);
// get data from depth sensor

depthFrame =  aligned.first(StreamType.COLOR).releaseWith(frameReleaser).as(Extension.VIDEO_FRAME);
// get data from rgb

If I do this in the same block will it give me the depth and RGB data at the same time?

bjain409, 2021-03-26:
> Hey @MartyG-RealSense I've looked through the source before to see how the renderer works but it uses opengl If I recall correctly which I'm not familliar with.
>
> However, I did manage to resolve the [...]

@Murded 

> FrameSet aligned = mPipeline.waitForFrames(1000).releaseWith(frameReleaser)
>                         .applyFilter(decimationFilter).releaseWith(frameReleaser) [...]

Hi @Murded 
Can you please share how did you create objects of 

`FrameSet aligned = mPipeline.waitForFrames(1000).releaseWith(frameReleaser)
                        .applyFilter(decimationFilter).releaseWith(frameReleaser)
                        .applyFilter(depthToDisparity).releaseWith(frameReleaser)
                        .applyFilter(spatialFilter).releaseWith(frameReleaser)
                        .applyFilter(temporalFilter).releaseWith(frameReleaser)
                        .applyFilter(disparityToDepth).releaseWith(frameReleaser);`

decimationFilter;
depthToDisparity
temporalFilter
spatialFilter
disparityToDepth
and even frameReleaser

And as mentioned its working for color camera
how would you capture for depth sensor

thanks.

bjain409, 2021-03-26:
> @kafan1986
> Thank you for responding back.
> I didn't completely follow this:
>
> ****" //Note: I Use a Mat here to calculate the size but you can easily calculate this manually but I don't have time to [...]

@kafan1986 
Can you please share the source how would I get the depth of individual points?

@Murded @MartyG-RealSense 
Please help with the requirements

bjain409, 2021-04-05:
> @kafan1986 I haven't read through all of your requirements but this should be more than enough to set you on the right track.
>
> How to get the depth buffer:
>
> ```
> //Get the frames from the camera [...]

@Murded Can you please share how can I get the depth of individual pixel/point?