realsenseai/librealsense #6090: T265: Level detection support

neilyoung, 2020-03-18:
Hi,

for my [PoC application](https://neilyoung.serveblog.net) I'm using for an effective way to support level detection (I mean floor level). Right now I don't see support for something like this in the SDK. My initial idea was to place different `static_nodes` at different levels, but this fails, since the SDK - once a scene is detected as "known" - automatically delivers the position of all nodes set on query. Initially I thought, they would only be delivered, if they come "in sight",  but the current behaviour makes totally sense.

In case I would use this, I could imagine to use the delivered Z distance as a level separator, but this seems a bit error prone to me.

Other ideas? Some way to mark a complete view as belonging to an outer context?

zainmehdi, 2020-03-19:
By floor level do you mean the floor height or like first floor second floor etc ?

neilyoung, 2020-03-19:
I meant the latter. Thinking twice I think a more general request would have to be the ability to label a scene. It is not only "level 1" or "level 2", it could also be "office room", "printer room" etc. Something which is implanted into a map at training time and returned at run time on recognition.  A "scene tag" so to say. But it might be too obscure, I suppose I will have to find a way outside the camera :) Don't bother with this...

radfordi, 2020-03-19:
I've tried to keep up aiming towards support for loading multiple maps with the idea being that you could connect to any or all of them.  This would allow a VR headset to detect which room it was in, for example.  Sadly, this not ready nor on the roadmap at this point in time.

neilyoung, 2020-03-20:
@radfordi That would be an awesome improvement, indeed... BTW: Have you ever seen my little PoC solution published these days? The results are that encouraging, that it literally cries for "Drive me forward" https://neilyoung.serveblog.net

RealSenseSupport, 2021-01-07:
Thank you again for highlighting these features. We have moved our focus to our next generation of products and consequently, we will not be addressing this in the T265.

closed 2026-10-23 by MartyG-RealSense