realsenseai/librealsense #6091: Applying postprocessing filters to bunch of points which are deprojected from image pixels by using rs2_deproject_pixel_to_point function

fburak, 2020-03-18:
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i} | 
| Firmware Version                   | (05.12.01.00) | 
| Operating System & Version |  Win 10 | 
| Platform                                 | Desktop PC  |
| SDK Version                            |  { 2.X }                          | 
| Language                            |  {C++/opencv/pcl/python}                          | 

I would like to have part of pointcloud which corresponds to part of color image taken from my D435i camera. For this reason I used rs2_deproject_pixel_to_point function with for loop and got 3d point representation of part of color image. Now I can generate pcl::pointcloud object and visualize them as point cloud. However, i would like to apply realsense postptocessing filters on this points generated with rs2_deproject_pixel_to_point function to increase quality. Postprocessing filters can only be applied on rs2::frame object. My question is how can i apply realsense postprocessing filters to the bunch of points generated with rs2_deproject_pixel_to_point function?

dorodnic, 2020-03-19:
You cannot do it. 
Obviously, post processing source code is open-source like everything else, but don't see how that's possible since the algorithms rely on local neighbor information

fburak, 2020-03-19:
Thank you for your response. @dorodnic 

As i explained in my question, I capture color image(from rs2::video_frame) and depth data(rs2::depth_frame) simultaneously. I do some image processing task and get a bounding box coordinates. I would like to get depth data (pcl::pointcloud, rs2::frame rs2::depth_frame or anything related to depth data) that corresponds part of image determined by bounding box coordinates of color image. My problem is similar to the problem in [this issue](https://github.com/IntelRealSense/librealsense/issues/3894). I want this obtained depth data to have good quality. I am working with live frames.

Firstly, I tried deprojecting points by using original rs2::depth_frame and color image. Then I was thinking about applying postprocessing filter to deprojected points to increase quality, but you said that it is impossible. 

I also tried to apply postprocessing filter to the rs2::depth_frame at the moment i captured. Then, I tried to obtain the point cloud by using the "rs2_deproject_pixel_to_point" method using the post processed rs2::depth_frame object. But here I got a strange set of points looks like to a pyramid.

Is there any other way to achieve this?

fburak, 2020-03-19:
Thank you for your response. @dorodnic 

As i explained in my question, I capture color image(from rs2::video_frame) and depth data(rs2::depth_frame) simultaneously and align them to color. I do some image processing task and get a bounding box coordinates. I would like to get depth data (pcl::pointcloud, rs2::frame rs2::depth_frame or anything related to depth data) that corresponds part of image determined by bounding box coordinates of color image. My problem is similar to the problem in [this issue](https://github.com/IntelRealSense/librealsense/issues/3894). I want this obtained depth data to have good quality. I am working with live frames.

Firstly, I tried deprojecting points by using color image pixels and distances and intrinsics of the original rs2::depth_frame object. Then I was thinking about applying postprocessing filter to deprojected points to increase quality, but you said that it is impossible. 

I also tried to apply postprocessing filter to the rs2::depth_frame at the moment i captured. Then, I tried to obtain the points by using the rs2_deproject_pixel_to_point method using color image pixels and distances and intrinsics of the **post processed rs2::depth_frame object**. But here I got a strange set of points looks like to a pyramid.

Is there any other way to achieve this?

dorodnic, 2020-03-19:
Using deproject with depth frame after post-processing should work. Could you share the code snippet used to deproject post-processed depth frame?

fburak, 2020-03-20:
I added my code and results. My code is modified version of the [rs-post-processing.cpp](https://github.com/IntelRealSense/librealsense/blob/master/examples/post-processing/rs-post-processing.cpp).
If you press space then it captures, process and saves data. If you press any other button it continues to render and show data. I aligned frameset in line 192, checked keypress in line 294, decided to use original frame or post processed frame in line 329 and 330, used rs2_deproject_pixel_to_point function in loop in line 345.

[code.zip](https://github.com/IntelRealSense/librealsense/files/4359168/code.zip)

[image]
[image]

RealSenseSupport, 2020-07-02:
@fburak Sorry for late reply. Is this still issue for you? Looking forward to your update. Thanks!

fburak, 2020-07-20:
> 
> 
> @fburak Sorry for late reply. Is this still issue for you? Looking forward to your update. Thanks!

yes, actually the problem continues and I am looking for a solution

RealSenseSupport, 2020-08-14:
I have tried to run your code with the latest librealsense(development branch). it seems the issue can't be reproduced in my side.
Would you try it again?

RealSenseSupport, 2020-09-04:
@fburak Did you get chance to try the latest librealsense? Looking forward to your update. Thanks!

RealSenseSupport, 2020-09-24:
@fburak Any update from your side? If no more questions we will close this in a few days. Thanks!

RealSenseSupport, 2020-10-10:
@fburak Sorry that we didn't hear from you for weeks. Close this accordingly. If you still have questions or issues, please feel free to create another new one. Thanks!

closed 2020-10-10 by RealSenseSupport