realsenseai/librealsense #6093: Announcement:  Highlighting RealSense ROS wrapper release 2.2.13

MartyG-RealSense, 2020-03-19:
Hi everyone,

I would like to highlight that there is a new release of the RealSense ROS wrapper (version 2.2.13).  It supports the following features:

* Librealsense SDK version 2.33.1
* The forthcoming new RealSense L515 lidar depth camera
* Simultaneous use of multiple T265 devices
* Kinetic and Melodic ROS distributions

The installation instruction for Melodic is:

**sudo apt-get install ros-melodic-realsense2-camera**

antoan, 2020-03-20:
I really appreciate this. What kernel would you recommend for Ubuntu 18.04 for a production system, stability?

MartyG-RealSense, 2020-03-20:
@antoan  You are very welcome!  The newest kernel supported in the RealSense SDK at the time of writing this is **5.3**.  If production stability is a priority then a good general rule of thumb is to use "one version back", whether it be SDK, firmware or kernel.  So the second newest kernel supported is Kernel **5**.

antoan, 2020-03-21:
Would very be grateful if you could recommend a way to 'hold' a HWE 5.0 kernel which came with Ubuntu 14.04.3 (if I understand things correctly,  the hardware enablement stack is necessary in order to get a 5.0 kernel on Ubuntu 18.04), to prevent it changing to the next point release kernel - 5.3, during upgrades. 

Was not able to find clear information on how to go about this elsewhere

MartyG-RealSense, 2020-03-21:
The link below describes a manual cloning process for HWE.

https://github.com/IntelRealSense/librealsense/issues/4464#issuecomment-513343764

closed 2020-03-27 by RealSenseCustomerSupport