realsenseai/librealsense #6094: Sensitivity curve Intel Realsense D435 (OV9282 and OV2740 sensors)

ToonTielen, 2020-03-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 

Dear support,
In order to make a good selection of my illumination source, I am searching for the sensitivity curves of the OV9282 (infrared) sensor and the OV2740 (color) sensor in an Intel Realsense D435. I did find the datasheets of Omnivision, but that gives no information about the spectral response of these sensors. Does anyone have additional information? @jkenney9a had the same question in #1070 but it was unanswered.

MartyG-RealSense, 2020-03-19:
I was asked this question myself a few months ago and was unable to find the curves.  I did some further research just now on the subject, and apparently that level of technical information from camera manufacturers requires a Non Disclosure Agreement (NDA) with that manufacturer to get access to.  So Intel would not be able to release the information publicly if they had it internally.

ToonTielen, 2020-03-20:
@MartyG-RealSense  thanks for the response. Should I have a NDA with Intel or with Omnivision to be able to access to that information?

MartyG-RealSense, 2020-03-20:
In order to be granted an NDA by Intel, they seek a Return On Investment (ROI).  An example of this is an economic one, such as intending to purchase a volume of more than 100,000 cameras.  

So it may be easier to seek one from OmniVision.  It may depend on how you intend to use the information though.  One person who sought such an NDA from them in 2013 reported that he had an initially positive response but was turned down once he disclosed that he wanted to use the information to create his own camera.

MartyG-RealSense, 2020-04-11:
This case will be closed after 7 days from the time of writing this message if there are no further comments.  Thanks!

ToonTielen, 2020-04-14:
Thanks for the information.

closed 2020-04-14 by ToonTielen