realsenseai/librealsense #6096: Get Point cloud as cv::Point3f

dbots94, 2020-03-19:
I just started using you library.

Searching for how to get a point cloud from the depth camera, I found an example that returns the set of points to a variable of type rs2::points.

But I was wondering if it is possible to return the point cloud set of points to a vector of cv::Point3f?

Best regards

MartyG-RealSense, 2020-03-19:
RealSense community member **UnaNancyOwen** has a Librealsense sample program for OpenCV that uses cv::Vec3f.  Would this meet your needs please?

https://github.com/UnaNancyOwen/RealSense2Sample/tree/master/sample/PointCloud

dbots94, 2020-03-19:
@MartyG-RealSense Despite the use of the type cv::Vec3f, I can't see where is the pointcloud being saved as a cv::Vec3f.

MartyG-RealSense, 2020-03-19:
I am still learning OpenCV, so I apologise if my reading of OpenCV code is rusty.  It looks as though the cv:Vec3f is being applied to a cv::mat through the cv:Vec3 vectors vertex.x, vertex.y and vertex.z.

https://github.com/UnaNancyOwen/RealSense2Sample/blob/master/sample/PointCloud/realsense.cpp#L221

And the vertex seems to have been set up earlier in the script by getting the points.

https://github.com/UnaNancyOwen/RealSense2Sample/blob/master/sample/PointCloud/realsense.cpp#L207

dbots94, 2020-03-20:
I understand what your saying but I can't figure out how to convert from rs::vertex to cv::Point whitout having to go through every point and converting it. It takes processing time that i need...

MartyG-RealSense, 2020-03-20:
The sense that I have is that to achieve what you are aiming for, you will need to look to a point cloud processing library outside of the RealSense SDK, such as PCL, as there is virtually nothing on cv::Point3f within the available references related directly to Librealsense code.

I located a non-English tutorial on cv::Point3f, OpenCV and PCL and auto-translated it into a PDF file.  I have attached it to this message in the hope that it can provide useful insights.

[OpenCV_Points.pdf](https://github.com/IntelRealSense/librealsense/files/4360378/OpenCV_Points.pdf)

dbots94, 2020-03-20:
Thank you for your help. One last question:

How do I create a cycle where I go through every point of the pointcloud, and acces its coordinates, so I can push_back the points coordinates to a vector of cv::Point3f?

MartyG-RealSense, 2020-03-20:
@dbots94    I researched your question very carefully.   If you are able to use PCL with OpenCV, the link below describes a method for converting a PCL cloud to Points3f using a "for" loop.

https://stackoverflow.com/questions/53781038/quick-conversion-of-pclpointcloudpointxyz-to-stdvectorcvpoint3f-using

dbots94, 2020-03-20:
That's just too much processing... So I'll have to convert from realsense pointcloud to pcl pointcloud, and then convert from pcl point cloud to std::vector<cv::Point3f>. All this conversions make my algorithm useless for realtime...

MartyG-RealSense, 2020-03-20:
If you want a simpler approach without having to use OpenCV, the RealSense SDK has a simple means in C++ of generating a point cloud with a processing block (pc.calculate) and get the vertices and texture coordinates.

https://dev.intelrealsense.com/docs/rs-pointcloud

dbots94, 2020-03-24:
@MartyG-RealSense Thanks for the help. That last suggestion helped

closed 2020-03-24 by dbots94

MartyG-RealSense, 2020-03-24:
Great news - thanks for updating us!   Good luck with your project.   :)