realsenseai/librealsense #6099: T265 No device connected

hamsterasu, 2020-03-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.926 | 
| Operating System & Version |    Linux (Ubuntu 16)  | 
| Kernel Version (Linux Only)    | 4.17.2                                        | 
| Platform                                 | Neutis N5 (arm64) |
| SDK Version                            |  2.33.1/2.33.2(tried both)                          | 
| Language                            |  python                          | 
| Segment			|  Robot                          | 

### Issue Description
Dear support, 
Previously I was able to run the T265 examples `rs-pose` and `librealsense/wrappers/python/examples/t265_example.py`  on my Neutis, an arm64 computer. But all of a sudden, I am getting errors as such when running rs-pose and rs-enumerate-devices:

Output of `rs-pose`:
```
RealSense error calling rs2_pipeline_start_with_config(pipe:0xb3dafd0, config:0xb3cc350):
    No device connected
```

Output of `rs-enumerate-devices`:
```
 20/03 14:40:04,527 ERROR [281472886112256] (tm-boot.h:39) Error booting T265
 20/03 14:40:06,619 ERROR [281472886112256] (tm-boot.h:39) Error booting T265
No device detected. Is it plugged in?
```

I have done fresh installations on new computers and tried it on different T265 cameras but I am getting the same results. The cameras are definitely working because I tried them on my laptop(x86) and I am able to run realsense-viewer, rs-pose and the other examples just fine. The funny thing is I can see the camera being detected by my Neutis when it is being connected in `dmesg` and `lsusb`. I have posted the relevant console outputs below. I am really at my wits end on this so I hope I can get some guidance on how to debug this problem. Thank you!

Relevant output of `dmesg` after disconnecting and reconnecting T265:
```
[11473.469108] usb 4-1: USB disconnect, device number 2
[11529.463436] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[11574.474568] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[11574.513350] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[11619.480170] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[11619.520269] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[11664.477874] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[11664.517165] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[11709.479673] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[11755.766116] usb 4-1: new high-speed USB device number 3 using ehci-platform
```

Output of `lsusb`:
```
Bus 008 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 003: ID 03e7:2150  
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 009 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

menandro, 2020-03-31:
Same thing for me. I rolled back to 2.31 for now.

hamsterasu, 2020-04-21:
Yes thank you! 2.31 worked for me. Tried 2.33 and 2.32 and they don't work.

closed 2020-04-21 by hamsterasu

robertocalvi, 2020-06-28:
For me librealsense 2.35.2 with Raspberry PI 4 4GB is not working at boot:

28/06 10:14:22,108 WARNING [3069497360] (messenger-libusb.cpp:66) bulk_transfer returned error, endpoint: 0x1, error: No data available, err. num: 61
28/06 10:14:22,108 ERROR [3069497360] (tm-boot.h:39) Error booting T265

If I unplug and replug USB works but only for a couple of minutes.

Could you help us?

hamsterasu, 2020-07-08:
Sorry, I'm not an expert at this. I can only suggest you to try a different sdk version like what I did above. If it does not work, then it is better that you open a new issue so that the Realsense developers will know your issue and assist you :)