realsenseai/librealsense #6100: After running pipeline.start (), will the camera continuously transmit frames to my computer, or will the camera only transmit frames to my computer when I call wait_for_frames ()?

Dontla, 2020-03-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435| 
| Firmware Version                   | 05.11.15.00| 
| Operating System & Version |  Win10  | 
| Platform                                 | PC  |
| SDK Version                            | 2.29.0                     | 
| Language                            |  python3.6.8                           | 
| Segment			|  Robot                       | 

After I enable pipeline.start (), if I don't call wait_for_frames (), will the camera consume my bandwidth?
Best regards!

MartyG-RealSense, 2020-03-20:
The purpose of wait_for_frames is to block until a frame is available.  It helps to ensure that a complete frame is received as soon as one is available.  It therefore provides stability by freeing up the CPU until a complete frame is available.

The meaning of wait_for_frames, poll_for_frames, etc and the differences between them is discussed in the link below.

https://github.com/IntelRealSense/librealsense/issues/2422#issuecomment-423254709

Dontla, 2020-03-21:
yeah, but if I don't call wait_for_frames (), would the camera consumes the USB bandwidth of my computer after  I had called pipeline.start ()?

MartyG-RealSense, 2020-03-21:
I would think that the main factors in bandwidth consumption are frames per second (FPS) and resolution, and use of wait_for_frames is more a matter of helping to reduce the likelihood of a single CPU core usage going to 109%.   

If you feel that you need some kind of frame mechanism in place but do not want a wait, follow Dorodnic's advice to use poll_for_frames (frames return instantly) and define when the CPU should sleep so the core does not always go to 100% usage.

dorodnic, 2020-03-22:
Hi @Dontla 
Frames are constantly streamed from the device in the background. They are only passed to you when you call `wait_for_frames`.

Dontla, 2020-03-23:
Yeah, my boss wants to connect 9 cameras to one computer, and I told him that according to the official suggestion, under a transmission configuration with a resolution of 1280 × 720 and 30 frames per second, a computer can only support up to 3 cams (https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration#section-2-multi-camera-considerations), but he thought that we don’t always use cameras at the same time, right? I guess he probably pointed the wait_for_frames (), indeed we only call wait_for_frames () for the next camera after we call wait_for_frames () for one camera, we do not use multithreading to call it, but I told him that even if we did not call wait_for_frames (), as long as we start the pipeline .start () of each camera, the frame set of each camera will be transmitted from the device to our computer through the USB cable constantly, but I was not sure if I am right, so I came here to ask you...

MartyG-RealSense, 2020-03-23:
@Dontla  Multiple camera setups do not always need to be hardware synced.  Multiple RealSense cam arrangements on the same computer were in use for years before hardware sync was available. 

In such systems,  the maximum typical recommended number of cameras active at the same time was 4.  Through a process of **enumeration**, tens of cameras could be cycled through in batches of 4 at.a time, taking a snapshot and then turning off those cameras and starting the next batch of 4, until you get back to camera number 1 and start the cycle again.

closed 2020-04-20 by MartyG-RealSense