realsenseai/librealsense #6101: Get depth map as CV_32FC1 opencv Mat

dbots94, 2020-03-20:
I'm using you library in parallel with OpenCV to get depth map, using c++ language.

Is there a way to get a cv::Mat of type CV_32FC1   (image with only one channel filled with double values) where that value is the distance to that pixel (in a known distance unit (meter, centimeter,...).

I'm already able to get the colorized depth map (RGB), but I wish to have a cv::Mat (GREY) where the pixel value is the distance from the camera to itself. 

Is there any way to get the depth map is this format? 

Is it possible to convert from the colorized frame to the grey distance map?

MartyG-RealSense, 2020-03-20:
OpenCV's documentation details a method for converting RGB to greyscale or greyscale to RGB.  Would this suit your needs please?

https://docs.opencv.org/3.4/de/d25/imgproc_color_conversions.html

dbots94, 2020-03-20:
No, that doesn't do what I asked for. I want to have a matrix with the distance between the camera and each pixel.

 You are only converting an RGB image to Gray. OpenCV does not know the mapping between the colors and the distance. So when converting, the returned image has its values mapped between 0 and 255, and those have no information about the distance.

dbots94, 2020-03-20:
That doesn't do what I asked. I want a Matrix filled with the distance between the camera and the respective pixel.

What you gave me is a conversion that has no information about the distance. When using that conversion, OpenCV doesn't know the correct mapping between the colors and the distance. So when doing that conversion, you'll get a matrix with values between 0 and 255, and those are not distance values.

MartyG-RealSense, 2020-03-20:
I will research your question further tomorrow and get back to you.   Thanks for your patience.

dbots94, 2020-03-20:
@MartyG-RealSense  thank you very much for all your replies!!

MartyG-RealSense, 2020-03-21:
@dbots94  I located a discussion with an OpenCV script that finds the distance from the camera to an object using a depth disparity map and CV_32FC1.  Is this closer to what you need please?

https://stackoverflow.com/questions/52182331/how-to-find-the-real-depth-distance-from-the-camera-to-the-object-in-m-from-di

ev-mp, 2020-03-22:
@dbots94 hello, probably one of the ways to achieve this is
```c++
    auto fs = pipe.wait_for_frames();
    if (auto depth_frame = fs.get_depth_frame())
    {
        auto pf = depth_frame.get_profile().as<video_stream_profile>();

        cv::Mat depth_metric_fp;
        cv::Mat depth_raw = cv::Mat(pf.height(), pf.width(), CV_16SC1, const_cast<void*>(depth_frame.get_data()));
        depth_raw.convertTo(depth_metric_fp, CV_32FC1);
        depth_metric_fp *= depth_frame.get_units();
        ...
    }
```

dbots94, 2020-03-23:
@ev-mp Hello, thanks for the reply, but I can't find the "get_units()" function...

dbots94, 2020-03-23:
@MartyG-RealSense Sorry, but I can't find any help in the link you sent. They are not using the same camera, neither the same library. I can't find where does that link relates to my question.

MartyG-RealSense, 2020-03-23:
@dbots94   The nature of OpenCV is that it is highly adaptable for different types of camera.  So code examples for RGB-D cameras (cameras with depth sensing capability) may also work with RealSense, so that they do not always have to be RealSense-specific examples in order to be useful.

ev-mp, 2020-03-23:
@dbots94 , `get_units()` from `rs2::depth_frame` [here](https://github.com/IntelRealSense/librealsense/blob/development/include/librealsense2/hpp/rs_frame.hpp#L822-L828) available from the `development` branch.

Note that in the above `auto depth_frame = fs.get_depth_frame()` the return type is `rs2::depth_frame` and not `rs2::frame`. 
If you sue this code make sure to downcast the frame handle in your code to the proper class 
`auto dpt_frm = orig_frame.as<depth_frame>();`

In case you're not building from source then you can get the depth metric conversion factor via the sensor API.

dbots94, 2020-03-23:
@MartyG-RealSense You're right. But my problem resides on the depth map capture. The example you gave is a construction of a depth map from a set of images already captured. In that example they load pgm images, and put them into cv::Mat. So there's no problems with types. The problem I have, is different because I'm receiving the image from a sensor, not from disc, so again, I can't see the relevance.

dbots94, 2020-03-23:
@ev-mp I followed the code you sent. Don't understand where you say I should use the `auto dpt_frm=orig_frame.as<depth_frame>();`
This is the code I have: 
`rs2::pipeline pipe;
pipe.start();
cv::Mat depth_metric_fp;

auto fs = pipe.wait_for_frames();
if (auto depth_frame = fs.get_depth_frame()){
   auto pf = depth_frame.get_profile().as<rs2::video_stream_profile>();
   cv::Mat depth_raw = cv::Mat(pf.height(), pf.width(), CV_16SC1, const_cast<void*>(depth_frame.get_data()));
   depth_raw.convertTo(depth_metric_fp, CV_32FC1);
   depth_metric_fp *= depth_frame.get_units();
}
cv::imshow("image", depth_metric_fp);`

`

ev-mp, 2020-03-24:
@dbots94 , that extra comment is for users that have their own mail loop defined.
Can you clarify whether the above code work correctly for you?

dbots94, 2020-03-24:
@ev-mp No that code did not work because the get_units() method is missing. I think I'm using the correct types

ev-mp, 2020-03-24:
@dbots94 , do you use code base from the `development` branch ?

dbots94, 2020-04-01:
@ev-mp I was using the `masters` branch

closed 2020-04-01 by dbots94