realsenseai/librealsense #6104: How do I improve the resolution of Realsense Viewer?

nbonwit, 2020-03-21:
I am imaging Lego bricks at about 200-300mm.  I want to get more resolution from the viewer.  I want to get sum-mm resolution (e.g., .1 or .01mm).

I have Controls->Depth units set to 0.0001 (i.e., 0.1mm).  I have Depth Table ->Depth units set to 1000.

When I hover the cursor over the depth map, it only shows 3 digits and the color map is very broad.  I want to visualize the bumps on the Lego bricks but don't know how to do it.
[image]

[image]

agrunnet, 2020-03-21:
You are adjusting the depth setting. You should be adjusting the “visualization” instead. There are adjustments for that in the viewer. You can set colors and ranges. 

Returning to the depth settings, I do want to add some comments. When viewing something near with the D415 (I presume that is what you have) you will benefit from reducing the depth units from default 1mm (1000um) to 0.1mm (setting of 100). And in order to get to this near distance you also need to do one or both of the following 1. Reduce resolution from 1280x720 to 480x240 which will reduce the MinZ from about 46cm to about 15cm. 2. Under advanced mode you change the disparity shift from zero to higher number.

agrunnet, 2020-03-21:
Regarding the on-screen readout in lower left of your image, the precision of that number is not adjustable in the app.

nbonwit, 2020-03-23:
Thanks for all your tips and information.  Regarding the visualization, this is how I have it set.  As far as I can tell, min/max only clips data but doesn't affect color ranges.  Do I have that correct?
[image]

Yes, I have a D415 camera.  If I set the Depth units to 100, then I get 0.1mm resolution, so if I set it to 10, then I get 0.01mm resolution?  Yes, I have been adjusting the disparity shift.

Also, why does reducing the resolution affect the MinZ?  What is the physics behind this?  I don't understand that part.  I would think that reducing the resolution would affect other things like spatial resolution, FOV, image transfer rate, etc.  So I don't understand how it would affect MinZ.

agrunnet, 2020-03-23:
The reason the resolution affects the MinZ is that we have a hard limit on the number pixels we search when compare left and right images. This limit is 126 pixels. But if there are fewer pixels in the image it means that you can effective search a  longer in terms of fraction of image. 

Yes the visualization knobs are the right ones. To see the effect you may have to change the color scheme. 

Regarding depth units, yes the smaller the number the smaller the depth steps that get reported. This is explained in more detail in the white paper talking about the best known methods for tuning performance. The paper also explains the MinZ.

nbonwit, 2020-03-23:
Thanks for explaining about MinZ and depth units.  I will review the tuning methods paper again on these points.

RealSenseCustomerSupport, 2020-04-02:
----------------------------------------------

Hi @nbonwit, do you have a chance to try agrunnet's suggestion and review the white paper? Do you need further help on this?
If we don't hear from you in 7 days, this issues will be closed.

closed 2020-04-16 by dorodnic