realsenseai/librealsense #6105: How do I run the factory calibration?

nbonwit, 2020-03-21:
I am measuring Lego bricks at close range (~200mm).  Unfortunately the D415 lenses are not focused properly at that range and are glued in place, so I cannot adjust them.

I have a small plano-convex lens in front of the camera that does a good job focusing the infrared camera image, but it messes up the depth map.

I tried recalibrating with the Calibration Tool, but it doesn't complete calibration with the lens in front, probably because the Cal tool only does extrinsic and not intrinsic parameters.

I want to run the factory calibration with the lens in front to update the intrinsic parameters.  I have searched and cannot find any tutorials on how to run the factory calibration.  Can you please get me started?

MartyG-RealSense, 2020-03-21:
Around 20 cm (200 mm) from the camera is the practical limit for minimum distance from an observed object on the 400 Series cameras.  Below that distance, the left and right IR sensors can become confused.

The free public version of the Dynamic Calibrator software calibrates only the extrinsics of the camera, whilst the OEM version of the software (aimed at engineering departments and manufacturing facilities) can calibrate both extrinsics and intrinsics.  

Unless you have an NDA agreement with Intel, the only other way I know of to obtain this OEM calibrator software is by purchasing the $1500 USD OEM calibration target system from the RealSense Store.

https://store.intelrealsense.com/buy-intel-realsense-d400-cameras-calibration-target.html

There may be options other than your plano-convex lens as a "cover material" for the camera's lens.  The subject of using a custom cover material to change the properties of incoming light is discussed at length here:

https://forums.intel.com/s/question/0D70P0000065HxqSAE

BTW, you may find useful a making-of article about a project that used a camera to isolate and identify Lego bricks moving beneath a camera along a makeshift conveyor.  

https://github.com/IntelRealSense/librealsense/issues/5393

agrunnet, 2020-03-21:
You will not be able to calibrate with a lens in front of each camera. Even if it works it would lose calibration almost immediately. This is why we build these units inside strong metal stiffeners. 
So what is right approach for you?
The D435 is meant for much closer range operation. The hyper focal length of the lenses is much shorter, the FOV is larger, and the minimum distance for depth at 848x480 is about 16cm. So it is ideal for your usage I think. 

By contrast the D415 has MinZ of about 45cm at 1280x720 so you would need to use disparity shifting. More importantly the image does go out of focus below about 45cm.

nbonwit, 2020-03-23:
Thank you for all your comments.  If I want to do an NDA with Intel, how do I set that up?  Who do I contact?

These things are obviously a roadblock for me, but perhaps the D435 will help.  I looked at the D435 originally, but when I compared it against the D415, it didn't seem as good as the D415.  Does it have lower resolution (I need highest possible resolution) or some other critical parameter?  I'm looking at the specs again now and it seems good, but I picked the D415 for some reason that I can't now remember...

MartyG-RealSense, 2020-03-23:
@nbonwit  To qualify for a RealSense NDA with Intel, the customer should typically offer a Return On Investment benefit to Intel.  A financial example of such a benefit would be an intention to purchase a volume of more than 100,000 cameras.  This is why the 1500 USD OEM calibration target board is the easier of the two ways to obtain the OEM calibrator software.

As @agrunnet said above, the next best step will likely be to switch to D435 and see how that improves your results when used without a lens in front of it.

nbonwit, 2020-03-24:
Thanks for all the info.  I'll give the D435 a shot when I have time....need to buy one first.

closed 2020-03-24 by nbonwit