realsenseai/librealsense #6106: InfraredRadiation image

fenneishi, 2020-03-22:
How can I get the InfraredRadiation images for D435 by SDK2.0?

agrunnet, 2020-03-22:
Yes it is the left imager.

MartyG-RealSense, 2020-03-22:
Hi @fenneishi  To access both left and right IR, if you visit the link below then there is a C++ script sample under the 'Librealsense2' heading.

https://github.com/IntelRealSense/librealsense/wiki/API-How-To#start-streaming-left-and-right-infrared-imagers

A C# version is here:

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/Documentation/cookbook.md#start-streaming-left-and-right-infrared-imagers

And one for Python:

https://github.com/IntelRealSense/librealsense/issues/1297#issuecomment-371418188

closed 2020-04-15 by MartyG-RealSense