realsenseai/librealsense #6107: Converting bag file into avi file for each stream like RGB and Depth Map

humayun, 2020-03-22:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I have bag files created using RealSenseD435 camera. I want to convert RGB and Depth image into avi or any other format which can input into Pose Estimation model. 

Second, these bag files are very big like in 3 to 5 GB for 1 to 2 minutes video. How can I crop object within the video and also decrease frame per second rate when converting into avi or any other format from bag file.

Thank you.

MartyG-RealSense, 2020-03-22:
The RealSense SDK has a conversion tool called **rs-convert** that can convert a bag file created in the SDK into a range of file formats (though not video formats such as avi).

https://github.com/IntelRealSense/librealsense/tree/master/tools/convert

A round-about way of creating a video would be to convert the bag into a set of PNG images with the rs-convert tools and then convert that image set into a video.  Googling for '**convert png set to video**' provides more information about ways to do this.

neilyoung, 2020-03-22:
You can derive the images from the bag and use ffmpeg to create a video again. Works at least for mp4. From mp4 to avi a conversion is possible.

closed 2020-04-15 by MartyG-RealSense