realsenseai/librealsense #6108: Motion Module failure

Gaoyuhanghaoren, 2020-03-22:
| | |
|---|---|
|**librealsense**|2.33.1 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435I|
|**Serial Number**|948122070824|
|**Firmware Version**|05.12.03.00|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|950323050691|
|**Firmware Update Id**|950323050691|

When I run the RealSense Viewer，it warning:Motion Module failure?
Is this caused by the hardware?

MartyG-RealSense, 2020-03-22:
The most recent discussion about motion module failure is at the link below:

https://github.com/IntelRealSense/librealsense/issues/6008

I am not certain of the exact cause of it yet.  Typically, the IMU data streams seem to continue to function as normal even when the error is being generated.

MartyG-RealSense, 2020-04-09:
This case will be closed after 7 days from the time of writing this message if there are no further comments.   Thanks!

closed 2020-04-15 by MartyG-RealSense

dlfksj, 2020-07-06:
I have the same issue.

When I run the realsense-viewer, it warns Motion Module failure.
And the same problem appears with ros2 wrapper. 
( It says : (ds5-motion.cpp:313) Motion Device is not calibrated! Motion Data Correction will not be available!)

How did you solve it?