realsenseai/librealsense #6111: When I manage to turn on Depth Sensor and RGB camera on Viewer, I meet this problem

jasonzhou123, 2020-03-22:
Hi,
When I turn on Depth Sensor and RGB camera at the same time. Both cameras shut down immediately and I got this report. By the way restarting the computer and installing a new Viewer made no sense.
"Backend in rs2_get_option(options:000001D3A13C4630, option:Backlight Compensation):
hr returned: HResult 0x800701b1: "指定不存在的设备。"
What's the problem?

MartyG-RealSense, 2020-03-22:
Could you provide details please of which operating system (Windows, Linux or MacOS) and which computer device you are using please?

dorodnic, 2020-03-22:
Also what type of camera

jasonzhou123, 2020-03-22:
Thanks! Operating system : Win 10. I use SR300.  The  USB  of my computer is is 3.0. And I am able to turn on Depth Sensor and RGB camera respectively but once I turn both of them on, I got error report as above.  Besides, my SR300 is  in good condition and I can use it smoothly on another Win 10 computer.

MartyG-RealSense, 2020-03-22:
Thanks so much for the additional information!   And which version number of the RealSense SDK are you using please?

dorodnic, 2020-03-22:
SR300 is known to have power surge on start, causing the camera to disconnect if the host PC is unable to provide sufficient power. This is consistent with the error you are seeing - 0x800701b1 is essentially "file not found". If you are using a laptop, please try when its plugged-in and not on battery mode. Also, powered USB hub is likely to solve the issue.

closed 2020-04-15 by MartyG-RealSense