realsenseai/librealsense #6112: D435i on Jetson Nano doesn't work with 2.32.1 from sources but works with 2.32.1 package

weblackey, 2020-03-22:
On a Jetson Nano with latest Jetpack updates, if we follow the Linux installation instructions and install  the apt packages then the D435i works in realsense-viewer. But if we compile the sources for the same version, then realsense-viewer cannot find anything attached. 

Sources compiled without any issue. udev rules are the same in both cases. Firmware on D435i has been updated. What could be the difference of the precompiled 2.31.1 vs the source code for the same release on github ?

weblackey, 2020-03-22:
OK this seems to be another instance of issue #6062

~/librealsense-2.32.1/build$ realsense-viewer
 22/03 18:24:38,861 WARNING [548396388800] (types.cpp:49) Accel Sensitivity:hwmon command 0x4f failed. Error type: No data to return (-21).
 22/03 18:24:58,347 WARNING [548387996096] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 25, 
 22/03 18:24:59,347 WARNING [548387996096] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 22, 
 22/03 18:25:00,415 WARNING [548387996096] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 22, 
 22/03 18:25:02,415 WARNING [548387996096] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 22, 
 22/03 18:25:02,643 WARNING [548387996096] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 25, 
 22/03 18:25:03,643 WARNING [548387996096] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 22, 
 22/03 18:25:04,523 WARNING [548387996096] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 11, 
 22/03 18:25:04,711 WARNING [548387996096] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 22, 
 22/03 18:25:06,387 WARNING [548387996096] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 25, 
 22/03 18:25:07,871 WARNING [548387996096] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 22, 
 22/03 18:25:10,011 WARNING [548387996096] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 22, 
 22/03 18:25:14,962 WARNING [548362817984] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 22/03 18:25:15,147 WARNING [548362817984] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 22/03 18:25:15,340 WARNING [548362817984] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 22/03 18:25:15,529 WARNING [548362817984] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 22/03 18:25:17,849 WARNING [548362817984] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 22/03 18:25:19,961 WARNING [548362817984] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 22/03 18:25:21,051 WARNING [548362817984] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 22/03 18:25:22,149 WARNING [548362817984] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 22/03 18:25:23,234 WARNING [548362817984] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 22/03 18:25:28,387 WARNING [547742544320] (messenger-libusb.cpp:105) usb_request_cancel returned error, endpoint: 130 error: Invalid argument, number: 22

weblackey, 2020-03-22:
ok this appears to be another

weblackey, 2020-03-22:
Sorry I meant to close the window, not the issue

In summary we have the issue that D435i works with released software 2.32.1, but does not work on Jetson Nano with the sources as distributed for the very same version 2.32.1 in all default settings. With special flags during compile it does work but still has a ton of error messages. So overall the D435i seems like a work in progress, more like an open beta test rather than a released product. 

And there seems to be some difference between the default settings for compiling the sources for 2.32.1 and the released apt packages for 2.32.1

dorodnic, 2020-03-23:
NVidia platforms pose certain challenges, and we are constantly working to address them. 
As we state in several places, D435i hardware and software was formally validated with Intel NUC hardware running Ubuntu 16 Kernel 4.4. Jetson Nano did not exist when D435i was originally released, so the only reason it works at all is due to continuous support and freely available software updates. We provide binary packages so you don't have to build from source. As mentioned in  #6062 we are looking into "Resource temporarily unavailable" issue on that platform.

weblackey, 2020-03-25:
Thank you for the background info. We only found out about the Realsense camera from the Nvidia reference designs so we didn't know the Jetson wasn't the primary platform. Yes we're trying it out on 18.04 and your binary release does work, both on Intel and Jetson platforms Is it possible to inquire about whether the Jetson binaries were compiled with special flags that are not the defaults for your source code release of the same version 2.32.1 ? Or maybe some additional bug fixes were merged in ?

RealSenseCustomerSupport, 2020-03-25:
----------------------------------------------

@weblackey According to https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md
the default build is set to produce the core shared object and unit-tests binaries in **Debug** mode. You can use `-DCMAKE_BUILD_TYPE=Release` to build in **Release** mode which used for pre-complied DKMS package. Did you build the source code in Debug or Release mode?

weblackey, 2020-03-25:
Thank you for the details. In compiling the sources we had followed the instructions and had used cmake ../ without any optional flags. That did not work. However, the flags mentioned in post #6062 did work: cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release

We have not narrowed it down further to see which of these made the difference but suspect it is the -DFORCE_RSUSB_BACKEND=true flag. But then if that wasn't set for the distributed binaries, it would be inconsistent with this conclusion and would suggest the Debug mode. We can try these flags individually to see what happens

dorodnic, 2020-03-25:
From Debian maintainer - `cmake .. -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=release -DBUILD_WITH_CUDA:BOOL=ON -DFORCE_RSUSB_BACKEND:BOOL=ON`

I'm dealing with the same issue on the Raspberry Pi, "Resource temporarily unavailable", and it seem to be highly non-deterministic. It may be something wrong we are doing in the code, or something to do with the Kernel version being used. I'm sorry for the issue, but we are actively looking into it.

weblackey, 2020-03-25:
Thank you for your comments.  We found that with the D435i these results are stubbornly deterministic. Maybe the D435i uses more of whatever resources are in question ? Or maybe it's a matter of what USB ID is being used. On the host, which is a Mac Pro, the keyboard and mouse use Bluetooth so the D435i is the only actual USB device we used in this experiment. On the Jetson Nano, the keyboard and mouse use USB so there are a minimum of 3 USB devices in use. 

We are scheduled to receive a D435 (without the i) today so we'll be able to figure out at least whether the D435i vs D435 difference factors in or not

weblackey, 2020-03-26:
Got the D435 today. Simply swapping out the D435i (with latest firmware) and connecting the D435 (with default firmware as shipped by Intel Realsense store) makes it work on Jetson Nano. We never tested the D435i with default firmware. The realsense-viewer had indicated that a newer version was available and recommended we went straight to the latest version. Now we're afraid of making any changes to the firmware version and will keep the D435 at the factory level that works.

RealSenseCustomerSupport, 2020-03-26:
----------------------------------------------

@weblackey What's the working firmware on D435? You can download the firmware from https://dev.intelrealsense.com/docs/firmware-releases and use RealSenseViewer to update the firmware.

weblackey, 2020-03-26:
The working D435 is using version 05.11.01.100 as shipped by Realsense store with no change.

The problematic D435i was using 05.12.03.00. Following your suggestion we used Realsense-viewer to change the firmware on the D435i to the same version  05.11.01.100 that works for the D435 and there was no difference.

In summary even with same firmware version, same cable, same Jetson Nano, same other attached devices, D435i still didn't work with compiled sources with no flags. It does work with your binary release while D435 works with both compiled sources and binary release.

The one thing we haven't tried yet is to compile sources with flags mentioned above but already it is clear that the D435i doesn't work under some conditions where the D435 works.

jorgemia, 2020-03-27:
@weblackey did the D435i work with binary release **with no flags**? I'm having some issues with mine with realsense v2.33.1. I'm getting warnings similar to the ones at the top of this issue frequently. Wondering if I should downgrade my librealsense and my camera?

dosvidos, 2020-03-27:
I have: 
D435i flashed with 5.11.15.0 firmware 
Jetson Nano running JetPack 4.3 L4T 32.3.1
The camera works fine with librealsense 2.17 built from source targeting the default RS2_USE_V4L2_BACKEND.
The camera doesn't work with librealsense 2.29 built from source targeting both the default RS2_USE_V4L2_BACKEND and alternative RS2_USE_LIBUVC_BACKEND backends.

weblackey, 2020-03-27:
@dosvidos  Yes that is consistent with what we're seeing on D435i on Jetson Nano and librealsense 2.29 built from sources without special flags. If you use librealsense sources 2.32 and then also use the following flags suggested by @dorodnic , @RealSenseCustomerSupport and debian maintainer, it may work for you, but you might see error messages with "resource unavailable"

cmake .. -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=release -DBUILD_WITH_CUDA:BOOL=ON -DFORCE_RSUSB_BACKEND:BOOL=ON

dosvidos, 2020-03-30:
Thank you, @weblackey 
BTW, one can use `cmake-gui` to see (and tweak) the full list of librealsense CMake variables available for a given platform.
The behavior observed with the recent versions of librealsense looks like a regression to my eyes, given that librealsense 2.17 works fine with the same HW.

RealSenseCustomerSupport, 2020-04-08:
----------------------------------------------

@weblackey So I understood the not working issue from source build has been resolved after set the customization flags, right? If so, can we close this issue now? Looking forward to your update. Thanks!

dosvidos, 2020-04-10:
For me, compiling librealsense 2.3x from source with CMake flags doesn't solve the issue of 435i camera being not detected on L4T 32.3.1 JetPack 4.3

weblackey, 2020-04-13:
Hi @MartyG-RealSense yes thank you. The solution was to use the custom CMAKE flags:

> the flags mentioned in post #6062 did work: 
> cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release

closed 2020-04-13 by weblackey

MartyG-RealSense, 2020-04-14:
@weblackey   I'm very pleased you found a solution that worked for you.  Good luck!

dosvidos, 2020-04-27:
Unfortunately, building librealsense 2.33.1 with FORCE_RSUSB_BACKEND CMAKE flag set to TRUE doesn't help in my case: RS 435i camera is not detected on L4T 32.3.1 JetPack 4.3

MartyG-RealSense, 2020-04-27:
@dosvidos  This case is closed.   Please start a new case by using the 'New Issue' button on the front page of the forum.   thanks!

https://github.com/IntelRealSense/librealsense/issues