realsenseai/librealsense #6113: Use Align function on a segment of the depth image

talbenach, 2020-03-23:
We need to find the depth of tags identified in the RGB image. Using the Align function on the entire IR image is resource consuming, and we are trying to reduce the burden of this process. Is it possible to define a range in the IR image and run the align function of that range alone (e.g. a range of 20X20 pixels)?

MartyG-RealSense, 2020-03-23:
In applications with a processing burden such as align, you can offload some of the work onto the GPU.  With Nvidia GPUs this can be done by building the SDK with CUDA support.   For non-Nvidia GPUs, vendor-neutral **GLSL Processing Blocks** can be used, though it may be ineffective on low-power devices.

https://github.com/IntelRealSense/librealsense/issues/4905#issuecomment-533854888

In regard to isolating part of the image, this could be done by defining a 'Bounding Box' area on the screen that excludes data outside of the box.  More information on this can be found here:

https://github.com/IntelRealSense/librealsense/issues/2016

talbenach, 2020-03-23:
Thank you very much MartiG. Can you expand a little on the bounding box option? In our application we will have 32 regions of interest for which we would like to map the depth values to the RGB image. Can you apply 32 bounding boxes to a single frame and run the Align function on them?

#4905 (comment)<https://eur01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FIntelRealSense%2Flibrealsense%2Fissues%2F4905%23issuecomment-533854888&data=01%7C01%7Cnoaml%40brainsgate.com%7C7b3d35c8425f4af80b1808d7cf0e9331%7Cba1b9a8211954a63b7107dab2a24ddaf%7C0&sdata=MWb3hFUibMW0thDY9DTyp09HoJHFFoL39D09KzlwwX0%3D&reserved=0>

#2016<https://eur01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FIntelRealSense%2Flibrealsense%2Fissues%2F2016&data=01%7C01%7Cnoaml%40brainsgate.com%7C7b3d35c8425f4af80b1808d7cf0e9331%7Cba1b9a8211954a63b7107dab2a24ddaf%7C0&sdata=C5d1VULC9mRGFR1kcYvk71vByjQCEzYLCG%2FerxOmnh8%3D&reserved=0>

MartyG-RealSense, 2020-03-23:
It is certainly possible to define multiple bounding boxes on-screen at the same time in real-time.  An example of this is the **MobileNet SSD** person tracking program, whose multiple-box ability is demonstrated in the link below.

https://github.com/apoorvavinod/Real_time_Object_detection_and_tracking#mobilenet_ssd-with-kcf-tracker

I would therefore speculate that 32 boxes on screen should be possible as long as your computing hardware can cope with the processing involved in keeping track of the individual boxes (though the calculations may be less with static regions of interest than with moving boxes).

As long as you do the align operation after the 32 boxes have been updated then I don't see why align would not work, as the align would probably just be doing a normal align of the image data that is left remaining after the bounding boxes have excluded the unwanted parts.

closed 2020-04-15 by MartyG-RealSense