realsenseai/librealsense #6115: (question) D435i Accelerometer Max Sample Time

JLBIZ, 2020-03-23:
### Info

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version |   Win10  | 
| Platform                                 | PC  |
| SDK Version                            |  2.33.1 }                          | 
| Language                            |  C++                        | 
| Segment			|  others                          | 

### Issue Description

Hello,
i am trying to **log acceleration data** from the D435i IMU.
For this purpose slightly modified the rs-motion sample project, in order to log the data in .csv Format. (I was not aware of the rs-data-collect project at that time).

`cfg.enable_stream(RS2_STREAM_ACCEL, RS2_FORMAT_MOTION_XYZ32F,fps);` is used to start the IMU Accel stream, and each frame is logged to .csv

According to my research, only **0**,**63** and **250** are valid values for the Accelerometer Framerate, whereas 0 means automatic selection of the framerate.

For my project, a sample rate of more than 250Hz would be really convenient. **Is there a way to achieve a sample rate faster than 250Hz??**

During my measurements i captured the following Acceleration data with fps=0 setting

> Timestamp[ms],AccX,AccY,AccZ
> 1584536421467.400879,0.402073,0.009807,10.846154
> 1584536421505.980957,-0.166713,-0.078453,10.218529
> 1584536421506.470947,-0.068647,0.000000,10.238142
> 1584536421506.958984,-0.166713,-0.039227,10.296982
> 1584536421507.446777,-0.137293,-0.029420,10.267562
> 1584536421507.934814,-0.078453,0.000000,10.296982
> 1584536421508.424805,-0.039227,0.000000,10.296982
> ...

In here, the time difference of each measurement is around 0.5 ms which would result in a sample_rate of **2000Hz**.
I am using the timestamp of the motion frame from the pipeline callback function.

After changing some settings, i could never reproduce the measurement with 2000Hz again.
**Is there any way to get the IMU stream with 2000Hz again??**

MartyG-RealSense, 2020-03-23:
A method that some users have employed to define their own custom IMU rates is to set up a timer.  One user created a RealSense ROS node in C++ that is controlled by a **rate** (they tried creating a node in Python but found it ineffective).  The link below provides an explanation about 'Rates' in ROS.

https://roboticsbackend.com/ros-rate-roscpy-roscpp/ 

Another user took the approach of setting up a timer in **QT Creator** instead of ROS.

https://doc.qt.io/archives/qt-4.8/timers.html

A different approach to custom rates that does not involve a timer is to set your program to only check every nth frame (e.g 1 in 5 frames) instead of every frame.

ev-mp, 2020-03-24:
@JLBIZ , with the current FW version only 63/250 Hz Accel rates are supported with the SDK.
Trying to modify the value manually can probably produce different rates, but in general results in undefined behavior. 

Can you elaborate on the use-case that you're developing?

JLBIZ, 2020-03-24:
@ev-mp Yes, when trying to set the stream to different values, we get an error.
When Polling the Frames without using the pipe callback, we can get more values but with the same timestamp, so currently no way to get faster than 250Hz.

What's confusing me are the above mentioned measurements with 2000Hz that i cannot reproduce at the moment. As timestamps and values differ, it seems to be valid measurement data.

Concerning the use-case:
We are already successfully using the RealSenseCamera on a machine for getting depth data and edge detection.
Now we also want to do vibration analysis on the machine. The idea was to use already existing hardware, e.g. the RealSenseCamera which has the IMU integrated.
But for serious vibration and frequency analysis, a signal sample_rate of at least 1000-2000Hz must be accomplished for this machine.
So if we can "tune" the RealSense to produce faster sample times, we could use it for vibration analysis as well, otherwise we must add another vibration sensor to our machine

JLBIZ, 2020-03-26:
The issue even becomes more and more strange.
I sent a compiled binary program to a colleague of mine.
The program only starts a stream via `cfg.enable_stream(RS2_STREAM_ACCEL, RS2_FORMAT_MOTION_XYZ32F,250);` and prints the motionFrame data to commandline.

While I am recording at **250Hz** my colleague is recording at **2000Hz**

- My measurements
> Timestamp[ms],AccX[m/s^2],AccY[m/s^2],AccZ[m/s^2]
> 1585212159003.772949,0.000000,10.159689,0.804145
> 1585212159005.784912,0.000000,8.855405,0.725692
> 1585212159009.750732,0.000000,10.042009,0.804145
> 1585212159013.716797,0.000000,10.218529,0.804145
> 1585212159017.682861,-0.009807,10.140076,0.804145
> 1585212159021.650879,-0.009807,10.140076,0.774725
> 1585212159025.618896,-0.009807,10.140076,0.804145

- My colleagues measurements
> Timestamp[s.ms],AccX,AccY,AccZ
> 1584536421467.400879,0.402073,0.009807,10.846154
> 1584536421505.980957,-0.166713,-0.078453,10.218529
> 1584536421506.470947,-0.068647,0.000000,10.238142
> 1584536421506.958984,-0.166713,-0.039227,10.296982
> 1584536421507.446777,-0.137293,-0.029420,10.267562

We are using the same Laptop model, same operating system, same usb-cable and same usb-port.
Firmware on my camera: `05.12.03.00`
Firmware on my colleagues camera: `05.11.01.100`

Just to be sure, i downgraded the firmware on my camera to `5.11.01.100`.
Still the same results.

**Any ideas how this completely different behavior could be explained?**

ev-mp, 2020-03-26:
@JLBIZ , producing 2000 fps rate for Accel technically viable as this rate supported by [the IMU](https://www.bosch-sensortec.com/products/motion-sensors/imus/bmi055.html#technical).

The reference you made about observing the data streaming at 2000 fps - it must have been a hw glitch and not part of D435i spec.

I'll forward the request for high rates for internal review and will update accordingly

JLBIZ, 2020-03-26:
Thanks a lot for your fast reply.
I checked out the datasheet of the IMU sensor as well.
If the RealSense is using unfiltered acceleration data, a sample rate of 2000fps seems possible(Datasheet page 24)
[image]

If this is a bug, it would be great to have this implemented as a feature, to get the 2000fps data from the IMU as well ;)

ev-mp, 2020-03-26:
The higher rate may affect bandwidth and CPU processing bottlenecks, especially with RaspberryPi-level boards and/or USB2 connection. 
As I mentioned, we'll take it for review.

MartyG-RealSense, 2020-06-30:
Hi @JLBIZ   Thank you very much for your patience as your feature request for support of a 2000 hz frequency was considered by Intel.  Unfortunately it was concluded that this feature could not be implemented.  Thanks again.  I will close this case now.

closed 2020-06-30 by MartyG-RealSense