realsenseai/librealsense #6116: Memory location error while trying to config

yaseenaamir, 2020-03-23:
well im using T265 to the pose data it throws some memory location exception while trying to config i have looked into the pose example and im using the same and im using 
the sdk version 2.21.0

[image]

neilyoung, 2020-03-23:
Out of curiosity: Is there a good reason to use such an "old" release? Current version is 2.33.1, IMHO

yaseenaamir, 2020-03-23:
oops sorry im using 2.33.1 ,by mistake i wrote that thanks for correcting

neilyoung, 2020-03-23:
I think it is hard to give you an advice on that, since your code is not known. As far as Python is concerned I can convince you, that there is no such a general problem as you describe. Even not with version 2.33.1. See also https://neilyoung.serveblog.net. My C is a bit rusty, but I should have means to check code, as long as it is broken down to the essentials. Try to roll a minimal project, maybe based upon the `rs-pose` sample and paste your crashing parts into it.

dorodnic, 2020-03-23:
Hi @yaseenaamir 
This just says that there is an error. If you catch exceptions in your C++ code you should be able to see what the actual error is, by calling `what`. I assume it's failing because Unreal sample project is designed with depth cameras in mind, and does not run on the T265 without modifications.

yaseenaamir, 2020-03-24:
@dorodnic i'm aware of that so i tried to modify and  get the data im attaching some sample of my code as well for better understanding
[image]

RealSenseSupport, 2021-01-07:
Hi @yaseenaamir 

Is there anything further you need on this?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2021-01-16 by RealSenseSupport