realsenseai/librealsense #6120: failed to set power state

soheilsbc, 2020-03-23:
I have tried to stream both my D415 cameras simultaneously. In the initialization, the first camera is getting initialized perfectly fine but the second camera gives me this error: failed to set power state
Any idea?

MartyG-RealSense, 2020-03-23:
There has been a similar still-open case recently with this error on a multi-camera setup on a discussion.  I recommend adding a comment there in order to collect the information of affected users together in one place.  Thanks!

https://github.com/IntelRealSense/librealsense/issues/5828

MartyG-RealSense, 2020-04-09:
This case will be closed after 7 days from the time of writing this message if there are no further comments.   Thanks!

closed 2020-04-15 by MartyG-RealSense

dheera, 2020-06-16:
I'm getting this issue as well.
[image]

dheera, 2020-06-16:
Also, firmware update results in Error type: Success (what?)
[image]

masynthetic, 2020-09-12:
@dheera in the same boat, with just one camera. Did you find a fix?

stino78, 2020-10-07:
I had a similar problem on Raspberry Pi OS, using pyrealsense2 and the D435i

I was opening two pipelines, one for the camera and one for the IMU. Starting the camera didn't give any problems, but when starting the IMU, right after, there was an error message "failed to set power state".

After may software and hardware attempts, I solved it in a trivial way... I added 500ms pause between the two starting.

You might start the first D415, wait 500ms and then open the second D415...

Good luck!

rythm07, 2021-11-17:
Hello @stino78, I have the same problem. How did you add the pause between the starting of the cameras?

stino78, 2021-11-17:
I just put a pause between the opening of the two pipelines in the code,
like half a second

rythm07, 2021-11-17:
Ah ok thanks @stino78, but I did not saw that you are using pyrealsense2. I'm using ROS so I think I have to find a different way

stino78, 2021-11-17:
I am not sure, but both ROS and pyrealsense should be wrappers of the same
C++ code, so the solution might be the same.

Also, the version of the SDK might matter A LOT.... at a certain point they
change the access method to the device, check that

Good luck!

rythm07, 2021-11-18:
oh ok i didn't know that. 
Thanks a lot for the insight, I will definitely check that!