realsenseai/librealsense #6121: Height accuracy for transparent objects

tispratik, 2020-03-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.12.03.00| 
| Operating System & Version |   Ubuntu 16.04 | 
| Kernel Version (Linux Only)    |  4.15.0-88-generic                                         | 
| Platform                                 |    - |
| SDK Version                            |  -                          | 
| Language                            |  python                         | 
| Segment			|  Robot                          | 

### Issue Description
Reference Image:
[image]

In the above reference image, we're getting a height of 7mm on the transparent section of the bottle and about 70mm on the wrapper portion. Is there a way we can do some kind of post processing on the depth data in the image to make the height data more uniform?

MartyG-RealSense, 2020-03-24:
@dorodnic has previously commented on the subject of transparent materials: "any material that is transparent in the visible spectrum should also be transparent to depth from stereo algorithm".

On objects that are difficult for the camera to see properly because of transparent or reflective properties, applying a professional scanning spray to them can make them more reliably visible to the camera.  Here is an example of such a spray:

https://3dscanspray.com/

A less pro method of achieving the same effect is to apply a fine spray-on powder such as baby powder or foot powder to the surface of an object.

If your project is not suited to applying a coating to the object then you could try aligning depth with color.  Intel take this approach in their **depth_under_water** Python tutorial on sensing through water and water filled transparent containers such as fish aquariums.

https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/depth_under_water.ipynb

tispratik, 2020-03-24:
Thank you for the pointers Marty. I will research the links. Earlier, when we applied the depth to color alignment, the FPS dropped by half.

MartyG-RealSense, 2020-03-24:
The additional processing involved in operations such as alignment may be partly offset by offloading some of the work onto the GPU.  This can be done by building Librealsense with CUDA support if you have an Nvidia GPU, or with the SDK's **GLSL** processing blocks if you do not have an Nvidia GPU.

https://github.com/IntelRealSense/librealsense/issues/4905#issuecomment-533854888

Vision computing operations can also be hardware-accelerated by inserting an **Intel Neural Compute Stick 2** device into a USB port alongside the camera.

https://store.intelrealsense.com/buy-intel-neural-compute-stick-2.html

tispratik, 2020-03-24:
Thanks for the information Marty. We are using a GPU, but we are reserving it for our deep neural network. I wonder how much of the GPU will be used by the Realsense camera? Can we have a GPU was well as a Neural compute stick... we're already having power issues powering two realsense cameras connected to our motherboard.

MartyG-RealSense, 2020-03-24:
The Neural Compute Stick 2 should not significantly drain resources.  Intel confirm in the store listing that it is a low-power device.

[image]

A VPU like the Movidius Myriad X one inside the Stick can carry out graphics processing independently onboard  without having to process it on the computer.

tispratik, 2020-03-24:
That helped immensely! With frame alignment and CUDA flag turned on, our FPS went back up from 7 to 20 FPS all the way to 29-30 FPS consistently. Thanks a lot!

MartyG-RealSense, 2020-03-24:
You're very welcome.  That's great news!   :)

tispratik, 2020-03-24:
So, the alignment wouldn't help much with the depth accuracy on transparent objects right? We cannot spray anything on the objects as we don't have control over them.

MartyG-RealSense, 2020-03-24:
I considered your question very carefully to think of possible approaches.  

If the label is the most consistently accurate part of the image, I wonder if you could define a 'bounding box' to focus only on the label part of the image and exclude the fully transparent parts of the bottle from the data by only having the label within the bounding box area.  It is a tricky subject though, especially if the bottles are moving past the camera in boxes along a conveyor.

Another approach might be to put some kind of 'tag' image on the bottle label that a camera can recognise and work out the position of, ignoring everything except the tag image's position.  On the T265 Tracking Camera this 'pose' of a tagged object can be detected with **Apriltags**, whilst 400 Series depth cameras may be able to detect a type of tag image called **Aruco**.  A tag-based approach would require less alignment precision than a bounding box.

Also, I wonder if the lidar-based depth of the new RealSense L515 may be able to provide better results with this transparency application than the stereo depth of the 400 Series.   @dorodnic  What do you think of that?

https://store.intelrealsense.com/buy-intel-realsense-lidar-camera-l515.html

tispratik, 2020-03-24:
Thanks for your quick response Marty. I am not sure of a good way to detect the labels on the bottles. Also, Aruco tags cannot be applied on the objects as they are not in our control. Yes, I am hoping that the lidar based camera will help, we had pre-ordered it when it came out.

MartyG-RealSense, 2020-03-24:
Another thought comes to mind.  If you know that the label is always going to be a certain distance from the camera, set a depth clamp (minimum and maximum allowed distance values).  Maybe 60 mm minimum, 80 mm maximum).  That should in theory exclude distances less than 60 mm or greater than 80 mm (or whatever min-max range you think best).

The Threshold Filter post-processing filter would be the appropriate way to do that.   It can be added to a program with C++, C# or Python scripting, or tested using the filter in the Post-Processing section of thee RealSense Viewer options side panel.

tispratik, 2020-03-24:
We had to disable the CUDA flag as the alignment got messed up. For some reason now we are not getting reliable alignment anymore, though we got the FPS we were expecting. This is affecting us adversely. Any thoughts?

tispratik, 2020-03-25:
We restarted the camera and the alignment is working again. Upto 22814 frames now for both color and depth, the difference in the frame numbers is 1 only. When the alignment is good, the difference is always 1. 

We will watch it, i fear that in a few hours the difference between the color frame number and depth frame numbers will start regressing. We have seen a difference of as high as 3 before the camera restart.

Current state as of this comment:
Color frame number: 22814
Depth frame number: 22813

MartyG-RealSense, 2020-03-25:
You could perhaps put a **hardware_reset()** instruction in your script to automatically reset the camera, and make its activation conditional on the frame difference being > 1.  Then the bottle scanning process could run without human intervention for manual camera resets.

tispratik, 2020-03-25:
Thanks Marty. The hardware reset sounds like a scary thing. What kind of attributes it resets? Can this cause camera's longevity/reliability to go down? Will it undo the Dynamic calibration? I read through the C++ code, but could not find any more information than "Send hardware reset request to the device."

neilyoung, 2020-03-25:
At least for the T265 a hardware reset doesn't damage the device. I'm using it by default in order to start from a defined state (unless I would like to re-use an accumulated map from inside the cam). I need to wait a second after a hardware reset in order to use the T265 again.

MartyG-RealSense, 2020-03-25:
Thanks @neilyoung   :)

@tispratik  A hardware_reset instruction has basically the same effect as physically unplugging the camera from the USB port and then re-inserting it.  It is very useful for industrial scenarios where the cameras need to be unmanned or are locked in a cabinet.  As @beilyoung said, it shouldn't harm the camera.

The outcome of a dynamic calibration is written to the camera, so a reset will not affect it.

tispratik, 2020-03-25:
Thank you Marty and Neil! Good to know about hardware_reset(). Is there a function which simply disconnects the camera and not reconnect automatically? For another use case I want to control when I disconnect and reconnect the remote camera.

neilyoung, 2020-03-25:
There have been some posts on a thread regarding a USB hub, which could control the power on its plugs. But the tread was about problems with re-enumerating the cams after the power cycle, so I'm not sure if this would be a solution.

MartyG-RealSense, 2020-03-25:
@tispratik  If you want to stop the camera streaming, you just stop the pipeline.  At the start of the script, the pipeline would have been defined.  For example:

**pipe = rs.pipeline()**

So to stop the stream, you do the opposite of pipe.start():

**pipe.stop()**

tispratik, 2020-03-25:
The use case is to disconnect the camera in order to do patch the linux kernel, once the patching is done, reconnect. Based on the instructions given on the linux kernel patching page.

MartyG-RealSense, 2020-03-25:
I would think that pipe.stop() would achieve that goal, as it would put the camera into non-streaming standby state, as if it had just been plugged in.  Because the pipeline would be inactive, this should allow the camera to be detected, patched and the pipeline then restarted with pipe.start()

MartyG-RealSense, 2020-04-15:
Closed due to no further comments.

closed 2020-04-15 by MartyG-RealSense

Aanalpatel99, 2022-04-15:
Hello,
I am new to Realsense, I have D435 and want to detect a glass door through the camera, however I am not able to find an appropriate solution for this, can you guys help me please.

MartyG-RealSense, 2022-04-15:
Hi @Aanalpatel99  Applying an inexpensive **linear polarization filter** sheet over the camera lenses on the outside of the camera can significantly negate glare from reflections.  An example of this is shown in the image below that demonstrates the improvement in glass visibility when using such a filter.

[image]

Aanalpatel99, 2022-04-15:
Thank you so much @MartyG-RealSense, I will try this.