realsenseai/librealsense #6122: How can I obtain the intrinsics of the depth camera (D435 Series) instead of color camera?

lanfeiying, 2020-03-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   | up-to-date | 
| Operating System & Version |   Archlinux  | 
| Kernel Version (Linux Only)    |  up-to-date                                        | 
| Platform                                 | PC  |
| SDK Version                            |  2                      | 
| Language                            |  C++/opencv                         | 
| Segment			|  Robot/VR/AR/others                           | 

### Issue Description
Hi,
I am wondering whether I can get the intrinsics of the depth camera in Realsense D400 Series.
I have read the documentation about the intrinsics. I only found it is feasible to obtain the intrinsics for video camera(colour camera). How can I get the intrinsics of depth camera?

The SDK provides the alignment of depth image and colour image. So, I think I can get intrinsics of depth camera. 
Is it possible if I follow the procedure of obtaining intrinsics of the video camera?
Regards

lanfeiying, 2020-03-24:
It seems that I can follow the procedure of finding intrinsics of the video camera. I will check it.

MartyG-RealSense, 2020-03-24:
The intrinsics and extrinsics of all sensors can be listed using a RealSense SDK program called **rs-enumerate-devices**.  

https://github.com/IntelRealSense/librealsense/tree/master/tools/enumerate-devices

The command to access the extrinsics and intrinsics is:

**rs-enumerate-devices -c**

dorodnic, 2020-03-24:
in code it would be something like:
```
pipeline p;
auto r = p.start();
auto intr = r.get_stream(RS2_STREAM_DEPTH).as<rs2::video_stream_profile>().get_intrinsics();
````

closed 2020-04-15 by MartyG-RealSense