realsenseai/librealsense #6123: No Realsense device found

abylikhsanov, 2020-03-24:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.12.02.100 | 
| Operating System & Version |   Ubunut Bionic  | 
| Kernel Version (Linux Only)    | 4.15.0-88-generic                                        | 
| Platform                                 | amd64  |
| SDK Version                            v2.33.1                
| Language                            | ROS/ROS2                       | 
| Segment			|  Robot                    | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Sometimes, when I restart the ROS node (note that I am wiritng an issue here as the problem is probably not with ROS wrapper but with the Realsense itself) the Realsense gets disconnected and I get warnings like:

`"No Realsense device was found!"`

Here are the few logs I could find from dmesg:

```
[    2.598006] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
[    2.598084] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[    2.598676] uvcvideo: Unable to create debugfs 2-3 directory.
[    2.598715] uvcvideo 2-3:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[    2.598717] uvcvideo 2-3:1.0: Entity type for entity Processing 2 was not initialized!
[    2.598719] uvcvideo 2-3:1.0: Entity type for entity Camera 1 was not initialized!
[    2.598792] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:08.1/0000:05:00.3/usb2/2-3/2-3:1.0/input/input4
[    2.598951] uvcvideo: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[    2.598957] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[    2.599219] uvcvideo: Unable to create debugfs 2-3 directory.
[    2.599262] uvcvideo 2-3:1.3: Entity type for entity Processing 7 was not initialized!
[    2.599264] uvcvideo 2-3:1.3: Entity type for entity Extension 8 was not initialized!
[    2.599267] uvcvideo 2-3:1.3: Entity type for entity Camera 6 was not initialized!

[  398.528140] traps: nodelet[1608] general protection ip:7fbe250d1dd0 sp:7fbdd27faf68 error:0 in libroscpp.so[7fbe2500e000+193000]
[  454.547480] usb 2-3: Process 2216 (nodelet) called USBDEVFS_CLEAR_HALT for active endpoint 0x82
[  456.578751] usb 2-3: Process 2216 (nodelet) called USBDEVFS_CLEAR_HALT for active endpoint 0x82
[  457.910255] usb 2-3: Process 2216 (nodelet) called USBDEVFS_CLEAR_HALT for active endpoint 0x82
[  476.377933] nodelet[2220]: segfault at 0 ip 0000000000000000 sp 00007f116f7fc758 error 14 in nodelet[556a1a409000+f000]
[  629.576946] traps: nodelet[2890] general protection ip:7f0e96e05dd0 sp:7f0e8da4df68 error:0 in libroscpp.so[7f0e96d42000+193000]

[  736.913242] xhci_hcd 0000:05:00.3: Timeout while waiting for setup device command
[  742.289126] xhci_hcd 0000:05:00.3: Timeout while waiting for setup device command
[  742.497109] usb 2-3: device not accepting address 3, error -62
[  742.937306] usb 2-3: Device not responding to setup address.
[  743.153176] usb 2-3: Device not responding to setup address.
[  743.361093] usb 2-3: device not accepting address 3, error -71
[  743.801174] usb 2-3: Device not responding to setup address.
[  744.017173] usb 2-3: Device not responding to setup address.
[  744.225082] usb 2-3: device not accepting address 3, error -71
[  744.665855] usb 2-3: Device not responding to setup address.
[  744.881727] usb 2-3: Device not responding to setup address.
[  745.089363] usb 2-3: device not accepting address 3, error -71
[  745.173407] usb 2-3: USB disconnect, device number 3
[  745.585656] usb 2-3: Device not responding to setup address.
[  745.801167] usb 2-3: Device not responding to setup address.
[  746.009393] usb 2-3: device not accepting address 4, error -71
[  746.449709] usb 2-3: Device not responding to setup address.
[  746.665707] usb 2-3: Device not responding to setup address.
[  746.873284] usb 2-3: device not accepting address 5, error -71
[  746.901335] usb usb2-port3: attempt power cycle
[  747.837703] usb 2-3: Device not responding to setup address.
[  748.053193] usb 2-3: Device not responding to setup address.
[  748.261277] usb 2-3: device not accepting address 6, error -71
[  748.701705] usb 2-3: Device not responding to setup address.
[  748.917786] usb 2-3: Device not responding to setup address.
[  749.125250] usb 2-3: device not accepting address 7, error -71
[  749.153039] usb usb2-port3: unable to enumerate USB device

```

ev-mp, 2020-03-24:
@abylikhsanov hello,
The error messages indicate that USB controller has issues to recognize and enumerate the device during the power-up/enumeration stage .
>[  744.881727] usb 2-3: Device not responding to setup address.
[  745.089363] usb 2-3: device not accepting address 3, error -71

Unfortunately most of those cases require user intervention:
-  HW setup: switching USB port, replacing/tightening USB cables.
- SW Setup: restart PC/cold boot.  You can also experiment with [rtcwake](https://linux.die.net/man/8/rtcwake) to temporally switch the system in deep sleep and check whether this could be enough to reset/troubleshoot USB controller.

weblackey, 2020-03-26:
It looks like we encountered the same issue in #6112

RealSenseCustomerSupport, 2020-03-30:
----------------------------------------------

@abylikhsanov Did you get chance to try what ev-mp suggested? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-04-15:
----------------------------------------------

@abylikhsanov Could you please update if you still need support on this?Thanks!

RealSenseCustomerSupport, 2020-04-22:
----------------------------------------------

@abylikhsanov Is the issue resolved? Looking forward to your update. Thanks!

dosvidos, 2020-04-22:
I to this date observe this issue with D435i connected to Jetson Nano running JetPack 4.3

RealSenseCustomerSupport, 2020-04-23:
----------------------------------------------

@dosvidos Did you try to use the custom CMAKE flags to see if it helps? Please refer to the discussion in https://github.com/IntelRealSense/librealsense/issues/6112#issuecomment-613149240

dosvidos, 2020-04-27:
Building librealsense 2.33.1 with **FORCE_RSUSB_BACKEND** CMAKE flag set to TRUE doesn't help in my case

RealSenseCustomerSupport, 2020-04-28:
----------------------------------------------

@dosvidos  Since your issue is not totally same as this issue, could you please create another new one to track? Thanks!

RealSenseCustomerSupport, 2020-04-28:
----------------------------------------------

@dosvidos Please also check if https://github.com/IntelRealSense/librealsense/issues/6293#issuecomment-619531747 can resolve your issue. Thanks!

ev-mp, 2020-05-04:
@dosvidos , in case you observe the same kernel log warnings then the above troubleshooting should apply as well.
>[  746.665707] usb 2-3: Device not responding to setup address.
[  746.873284] usb 2-3: device not accepting address 5, error -71

Otherwise - please follow @RealSenseCustomerSupport  suggestion and open a dedicated ticket.

closed 2020-05-04 by ev-mp