realsenseai/librealsense #6124: RFE: Move python cmake target into its own namespace

morxa, 2020-03-24:
Currently, the python bindings are installed into the same namespace as the main library. This is problematic for packaging on Fedora: Recently, we have added the python bindings and put them into a separate sub-package (`python3-librealsense` and `python3-librealsense-devel`). However, it is no longer possible to develop against librealsense (with `librealsense-devel` installed) without also having the python bindings installed. This is because the cmake target checks that all files are available. If the python bindings are packaged separately, then those files are missing if the package is not installed, resulting in errors as follows:
```
$ cmake ..
CMake Error at /usr/lib64/cmake/realsense2/realsense2Targets.cmake:89 (message):
  The imported target "realsense2::pyrealsense2" references the file

     "/usr/lib64/pyrealsense2.cpython-38-x86_64-linux-gnu.so.2.33.1"

  but this file does not exist.  Possible reasons include:

  * The file was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and contained

     "/usr/lib64/cmake/realsense2/realsense2Targets.cmake"

  but not all the files it references.

Call Stack (most recent call first):
  /usr/lib64/cmake/realsense2/realsense2Config.cmake:45 (include)
  CMakeLists.txt:6 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/thofmann/test/librealsense-cmake/build/CMakeFiles/CMakeOutput.log".
```

This problem could be solved by moving the python bindings into a separate namespace, similar to `librealsense-gl`. However, I do not want to do this downstream, as it would deviate too much from upstream (i.e., your) naming conventions.

Would you be interested in such a split? If yes, I'd be happy to submit a PR.

dorodnic, 2020-03-24:
Hi @morxa 
We are always interested and open for community contributions.
For Debian eco-system, as far as I know, we do not include python binaries in Debian packages, but rather publish PIP wheels instead.

RealSenseCustomerSupport, 2020-04-08:
----------------------------------------------

@morxa Did you get chance to submit your PR? Please link to this when you finish. Thanks!

morxa, 2020-04-08:
Thanks for the follow-up! I have been working on it but I haven't been able to finish it yet.

RealSenseCustomerSupport, 2020-04-21:
----------------------------------------------

@morxa Any PR progress from your side? Thanks!

morxa, 2020-04-21:
@RealSenseCustomerSupport please have a look at #6220.

RealSenseCustomerSupport, 2020-04-22:
----------------------------------------------

@morxa Thanks very much for the contribution! I can see our engineers are looking at PR#6220. Let's wait to see further update there. Thanks!

RealSenseCustomerSupport, 2020-05-19:
----------------------------------------------

@morxa Can we track this with PR#6220 https://github.com/IntelRealSense/librealsense/pull/6220 and close this ticket? Thanks!

morxa, 2020-05-19:
Usually, issues are closed when the respective PR is closed. But it's not my repository, so feel free to do whatever you prefer.

RealSenseCustomerSupport, 2020-05-29:
----------------------------------------------

@morxa Thanks for the reply! Let's keep the ticket open till PR closed.

RealSenseSupport, 2020-07-02:
@morxa [PR#6220](https://github.com/IntelRealSense/librealsense/pull/6220) now merged. Can we closed this accordingly? Looking forward to your reply. Thanks!

PR #6220 (2020-06-22): Move python bindings into separate cmake target
* Move python bindings into a separate cmake target so we can install them separately (see #6124).
* Also install the python bindings into python's sitearch, not in `/usr/lib64` (or `/usr/local/lib64`).

Resolves #6124.
closed 2020-07-02 by morxa