realsenseai/librealsense #6126: Support Python3.8

zwn, 2020-03-24:
Please add precompiled wheels for python3.8 to pypi.

MartyG-RealSense, 2020-03-25:
Hi @zwn   Your message has been tagged by @dorodnic the RealSense SDK manager with the 'enhancement' label that is applied to feature requests, so that the RealSense team can officially track the request.  Thanks so much for your input!

zwn, 2020-03-26:
It would be really great if python3.8 wheels could be added also for the release 2.31 since that is the last release where t265 can be opened by serial number or can work alongside d435i. That is the release we are using (I've just tested 2.33 and the behavior is the same as 2.32).

MartyG-RealSense, 2020-03-26:
@zwn  As well as searching by serial number, you can also look for the first device with the word 'T265' in their name.

https://github.com/IntelRealSense/librealsense/issues/3434#issuecomment-476290706

ROS users also have the option of using the ROS wrapper to identify devices by USB port ID instead of serial number with **usb_port_id**:

https://github.com/IntelRealSense/realsense-ros#launch-parameters

zwn, 2020-03-26:
Actually, the blocker is that we cannot use t265 and d435i together at all (with or without serial number). The pipeline for t265 fails to resolve with 2.32 and 2.33 if d435i is plugged in :worried:. It is the simplest possible pipeline:

```python
self.pose_pipeline = rs.pipeline(ctx)
pose_cfg = rs.config()
pose_cfg.enable_stream(rs.stream.pose)
self.pose_pipeline.start(pose_cfg, self.pose_callback)
```

If I unplug the d435i, it works fine. It is unfortunate to have such a big regression but I appreciate the work being done to move the whole project forward. We just stay with 2.31 for the time being.

MartyG-RealSense, 2020-03-26:
I've added a 'T260' label to this discussion to help inform RealSense support team members that your question involves T265.

MartyG-RealSense, 2020-06-14:
@zwn  I have officially logged with Intel the feature request for Python 3.8 support for PyPi installation.  Thank you very much.

tomgoddard, 2021-02-09:
Python 3.9 and 3.8 have been released and yet pyrealsense2 on PyPi only offers Python 3.7.  It appears Python support for RealSense is dead.

MartyG-RealSense, 2021-02-09:
Hi @tomgoddard   Python support for RealSense has not been abandoned.  

In the link below, an explanation is offered for the time taken to implement support for Python versions newer than 3.7.

https://github.com/IntelRealSense/librealsense/issues/8229#issuecomment-767348365

A method for hex-editing the Python 3.7 pip **whl** file was also suggested in that discussion by a RealSense community member.

https://github.com/IntelRealSense/librealsense/issues/8229#issuecomment-767115385

MartyG-RealSense, 2021-03-21:
Hi everyone,

I wanted to update you on progress.  SDK version **2.43.0** has just been released and has the following comment in its release notes:

**[Python] Generate wheels for Python 3.8 + 3.9   Package classifiers updated to support python 3.8 + 3.9** 

[image]

MartyG-RealSense, 2021-03-21:
Hi @zwn   This case should be kept open until it is confirmed that the Intel feature request (**Enhancement**) associated with it is concluded.  Thanks!

zwn, 2021-03-21:
@MartyG-RealSense It is confusing. Commonly, the issue is closed when the reporter is happy. Where is the Intel feature request that is reportedly still open? Why is it still open? Why doesn't Intel use this tracker? Never mind.

MartyG-RealSense, 2021-03-21:
Yes, I understand.  I apologize for the confusion.  It is usually true that a case can be closed by the customer when they are satisfied.  Cases that are linked to an official Change Request (indicated by an **Enhancement** tag on the case) have to be kept open until that request is concluded though.  It is an internal Intel tracker, so you had no way to know and did nothing wrong at all.

MartyG-RealSense, 2021-03-31:
Hi @zwn   As the **pip install pyrealsense2** wheels for Python 3.8 and 3.9 are now available and the Intel feature request has been concluded, this case can now officially be closed.  Thanks again!

closed 2021-03-31 by MartyG-RealSense