realsenseai/librealsense #6127: OpenVINO "Face" Sample Fails on Windows with OpenVINO 2020.1

RMichaelPickering, 2020-03-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415 | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC  |
| SDK Version                            |  2.33.1                          | 
| Language                            |  C++/OpenVINO }                          | 
| Segment			|  {others }                          | 

### Issue Description
The supplied "Face" sample for OpenVINO fails to run on Windows 10 when built with the latest production release, 2020.1, of OpenVINO. We have verified that it works as documented when built with OpenVINO 2019.3.

RMichaelPickering, 2020-03-25:
NOTE: This issue appears to be due to breaking changes implemented with the latest OpenVINO release 2020.1. In particular, the solution itself doesn't get generated successfully by Cmake because a .hpp file required from OpenVINO installation folders has been moved. Even when we correct that issue, the build itself failed. This appears to be because the OpenVINO inference engine library has changed with release 2020.1.

dorodnic, 2020-03-25:
Hi @RMichaelPickering 
We are aware of this issue and considering how to deal with different versions of OpenVINO

RMichaelPickering, 2020-03-25:
Hi @dorodnic 
Good to know this! At the same time, I hope that you're aware that one compelling reason for this sample to work with OpenVINO release 2020.1 is because the OpenVINO Deployment Manager is supposed to work to allow generating a standalone deployment of the application on Windows. However, in our testing so far we haven't managed to get it to work on Windows at all. Thus, given that the same sample already works with the previous OpenVINO 2019.3 release, I don't see it as necessary to spend much effort updating the sample to work with OpenVINO 2020.1 until such time as some fix(es) are available such that the Deployment Manager works for deploying to Windows. Until such time, I'd suggest that it would be sufficient to make a note that the sample is only supported with OpenVINO 2019.3.

RMichaelPickering, 2020-04-23:
@dorodnic Just bumping this issue after the release of OpenVINO 2020.2. Given the extent of the performance improvements and additional models now supported with this release, it now seems that it is crucial for the RealSense samples to work with this release, too! Any chance that there will be an update for the RealSense OpenVINO wrapper and/or samples so that they work with 2020.2?

RMichaelPickering, 2020-04-30:
@dorodnic Is there any update on this issue or a roadmap for updating the RealSense OpenVINO samples and wrapper to use OpenVINO 2020.2 codebase?

RMichaelPickering, 2020-05-12:
@dorodnic @RealSenseSupport Should I close this issue and open a new one regarding OpenVINO 2020.2 instead of 2020.1? The point is that the priority for fixing the sample as well as the included inferencing capability for the VIewer is to make both work with 2020.2 rather than 2020.1 now!

RMichaelPickering, 2020-06-16:
@RealSenseCustomerSupport @RealSense-Customer-Engineering Can we please get this assigned to a developer? OpenVINO 2020.3 has been delivered and it is now designated as an official LTS release!!

mgomesborges, 2020-08-03:
@RealSenseCustomerSupport @RealSense-Customer-Engineering @dorodnic @RMichaelPickering 

I have been trying to compile LibRealSense + OpenVINO on Windows, macOS, Linux, and Raspberry Pi. But unfortunately, I couldn't. I confess that I am frustrated and disappointed.

I have the impression that the documentation is out of date. Some tutorials are missing details like the LibRealSense version and  OpenVINO version for which the tutorial can guarantee that it compiles. This lack of information has generated several messages in the issues section, and it seems to me that it is better to read the issues than to read the documentation itself.

I believe I can help you update the docs, and fix some bugs. I will send some pull requests as soon as possible. Maybe Intel can hire me to work one day a week to help you guys with those challenges.

Anyway, if you can share some secrets and some additional information to solve the LibRealSense + OpenVINO compilation problems, everyone will be grateful.

RMichaelPickering, 2020-08-03:
@mgomesborges Thanks for asking about LibRealSense and OpenVINO! I certainly understand your frustration, as it has now been more than four months that my issue has been open, and still the latest version of the RealSense SDK must be built with a version of OpenVINO from 2019 for it to fully work -- but of course that doesn't give us access to the latest features and performance improvements available in OpenVINO!!
I am beginning to think that @RealSenseSupport is, like many of us, hoping that 2020 will just go away...

RealSenseSupport, 2020-08-10:
Hi,

This is a known issue and are planning an SDK update that supports newer versions of OpenVino as soon as possible.

Thanks for your patience

mgomesborges, 2020-08-13:
> Hi,
> 
> This is a known issue and are planning an SDK update that supports newer versions of OpenVino as soon as possible.
> 
> Thanks for your patience

@RealSenseSupport @dorodnic You can count on me to run tests or help you fix this problem.

RealSenseSupport, 2020-12-29:
Hi,

This should now be resolved as of this release:

https://github.com/IntelRealSense/librealsense/releases/tag/v2.41.0

Please let us know if you will be needing further assistance with this issue.

Thanks

RMichaelPickering, 2020-12-29:
@RealSenseSupport Could you please clarify if OpenVINO 2021.2 will also be supported? Though I'm aware that it's still 2020, the latest releases of OpenVINO are numbered 2021.1 and 2021.2!

RealSenseSupport, 2021-01-11:
Hi,

As of the v2.41.0 release all 2020 versions of OpenVINO should work. While future versions of OpenVINO may work with our SDK, the possibility of breaking changes being introduced always exists, so it's still best to use supported/LTS version(s) of OpenVINO for maximum compatibility..

Thanks

closed 2021-03-27 by RealSenseSupport