realsenseai/librealsense #6128: Labview wrapper version

Viraxt, 2020-03-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | up-to-date | 
| Operating System & Version |  Win 10 | 
| Kernel Version (Linux Only)    |                                        | 
| Platform                                 | PC  |
| SDK Version                            |   2                          | 
| Language                            |  LabVIEW                   | 
| Segment			|                        | 

Hello everyone,

I just downloaded the library to use the camera with LabVIEW. The issue is I can't open the VIs because I need LabVIEW 2017 at least to open it and I only have the 2015 version.

Is there somewhere I can find a 2015 version, or someone can "downgrade" me the library so I can use it ?
Thanks

agrunnet, 2020-04-30:
Please try this:
http://realsense-hw-public.s3-eu-west-1.amazonaws.com/Releases/RS4xx/Windows/Intel%20RealSense%20for%20LabVIEW%202015.zip

RealSenseCustomerSupport, 2020-05-06:
----------------------------------------------

Hi @Viraxt Did you have a chance to try this?
If we don't hear from you in 7 days, this issue will be closed.

Viraxt, 2020-05-06:
Sorry I didn't saw the first reply. 

The attached zip from agrunnet works well, I can open the VIs with my 2015 LabVIEW as I wanted, thank you.

I think that you can close the topic.

closed 2020-05-11 by Viraxt