realsenseai/librealsense #6129: T265 API with wheel odometry error

Geonhee-LEE, 2020-03-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265  | 
| Firmware Version                   | 0.2.0.926 | 
| Operating System & Version |   Linux  16   | 
| Kernel Version (Linux Only)    |  4.15.0                                        | 
| Platform                                 | PC |
| SDK Version                            |    2.33.1                           | 
| Language                            |  C/python                         | 
| Segment			|  Robot                          | 

### Issue Description

Hello, I'm using T265 for integrating with mobile robot.
I have been searching many materials and videos, but I have some problems. Please to reply for my fault. thanks.

First of all, my goal will achieve nice odometry using T265. According to [description](https://github.com/IntelRealSense/realsense-ros#using-t265), I should integrate with wheel odometry. When I referred the many things following:
- https://github.com/IntelRealSense/realsense-ros#using-t265
- https://github.com/intel-ros/realsense#using-t265 
- etc

It isn't helpful for me, calibration json file only describes the fixed coordinate which mobile robot forward direction is z(don't have another examples). But, my robot odometry direction is x(red).  In this case, how to set [the calibration odometry file](https://github.com/IntelRealSense/librealsense/blob/master/unit-tests/resources/calibration_odometry.json)?
When I follow the TF convention in the ROS, it seems like fail..

Second, I could run examples of librealsense without `[t265_wheel_odometry.py](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/t265_wheel_odometry/t265_wheel_odometry.py)` of t265_wheel_odometry folder. Anybody run this python file?  I found that the 46 line occur error in the `t265_wheel_odometry.py`

`Traceback (most recent call last):
  File "t265_wheel_odometry.py", line 46, in <module>
    pipe.start()
RuntimeError: No device connected
`
I already connected the T265 camera and successfully ran another examples.

Third, USB connection error sometimes occurs.. 

> [ERROR] [1584930529.500434444]: The requested device with device name containing t265 is NOT found. Will Try again. 

I know the usb port support enough power because I referred the datasheet. But, I used the USB port which can support 10W more. What is the problem? Can I solve it?

Finally, why T265 imu topic doesn't exist orientation value? T265 imu topic just has accelerate values. In order to fuse the filter, I need orientation values.

Thank you. Please understand my bad English skill.

neilyoung, 2020-03-25:
You should start with the essentials first. Then you might think about wheel odometry (what nobody needs, IMHO). I would suggest you to try my solution first https://neilyoung.serveblog.net/

> I know the usb port must support enough power because I referred the datasheet. But, I used the USB port which can support 10W more. What is the problem? Can I solve it?

The power consumption on a 5 V supply is about 150 mW. 

> Traceback (most recent call last): File "t265_wheel_odometry.py", line 46, in <module> pipe.start() RuntimeError: No device connected

This is also sometimes the response to a crashed device. In that case only a power cycle of the Camera helps.

>Finally, why T265 imu topic doesn't exist orientation value? T265 imu topic just has accelerate values. In order to fuse the filter, I need orientation values.

What do you mean by that? The T265 delivers orientation by providing translation and rotation. What are you missing?

Geonhee-LEE, 2020-03-26:
@neilyoung  Hello, thanks to your reply.

I entered the your [GIthub repo](https://github.com/neilyoung/t265pose/tree/master), any python file doesn't exist. Can you share me with your code? Anyway, your blog is really arranged well, thanks.

I know the T265 needs about 2W at least. My on-board PC can supply power more than 2W , but it isn't worked strangely. 

I mean [T265 IMU ros topic](http://docs.ros.org/melodic/api/sensor_msgs/html/msg/Imu.html) don't include orientation value and only include velocity and acceleration value.So as your recommendation, can I use only odom topic which is provided translation and rotation to use orientation?

Thanks

neilyoung, 2020-03-26:
The code is in the release section. Current release is 0.6.0.

lbxxdi, 2020-09-23:
Hi @Geonhee-LEE 

I found this error too. and fixed by

Insert
`cfg.enable_stream(rs.stream.pose)`
Below
`cfg = rs.config()`

And 
change 
`pipe.start()`
to 
`pipe.start(cfg)`

closed 2026-10-23 by MartyG-RealSense