realsenseai/librealsense #6130: get_frame_number() returns strange values

patmalcolm91, 2020-03-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version |   Linux Mint 19.3  | 
| Kernel Version (Linux Only)    |                       5.3.0-42-generic                 | 
| Platform                                 | PC |
| SDK Version                            | 2.33.1-0~realsense0.2079                        | 
| Language                            |  python          | 
| Segment			|  other       | 

### Issue Description
I'm running a simple python script which consists of a loop in which I call ```pipeline.wait_for_frames()``` (color and depth streams enabled), and do some processing on the frame data. I'm attempting now to use the frame number for some other checks, but I'm noticing that the values returned from ```get_frame_number()``` are very strange. Here is a sample:
```
color frame number, depth frame number
0 18
2 20
5 23
2 24
4 26
5 27
6 28
8 30
9 31
10 32
```

Both streams are configured at 30 fps, and if I manually count iterations in the loop and time the process, it comes to very nearly 30 fps, but the frame numbers shown do not appear to follow any pattern (sometimes they even decrease from previous iterations, sometimes repeat, and sometimes jump by a large amount).

Any help or idea on why this is happening would be greatly appreciated.

MartyG-RealSense, 2020-03-25:
Although FPS can be set to a fixed value, it does not necessarily mean that a stream will constantly run at this speed.  It may vary depending on environmental factors (e.g changes in lighting conditions).  The FPS speed can be enforced at a constant rate by having Auto-Exposure enabled but disabling a setting called Auto-Exposure Priority (found in the RGB section of the side-panel options in the RealSense Viewer).

Regarding anomalies in frame numbers: if the stream "hiccups" so that it does not generate a good frame, it may return to the last good frame before proceeding onward.  The subject of hiccups and frame jumps is discussed in the link below, where a member of the RealSense team provides advice.

https://support.intelrealsense.com/hc/en-us/community/posts/360031847974-Locating-end-of-bag-file-poll-for-frame

patmalcolm91, 2020-03-25:
Thanks for the quick response, @MartyG-RealSense. The slightly varying framerate isn't a big concern for me, but it's good to know that it can be fixed by turning off auto-exposure priority.

The hiccups and frame jumps explanation is what I was looking for. I'm still unclear on how one could possibly get the pattern I'm seeing at the beginning (0, 2, 5, 2, 4, 5) if the camera is just grabbing the last good frame (surely it would then be 0, 2, 5, 5, 5, 5), but for my purposes I don't think ```get_frame_number()```  will be useful for what I had in mind anyway, given the way it appears to work. Thanks again for the fast answer!

closed 2020-03-25 by patmalcolm91

ev-mp, 2020-03-25:
@patmalcolm91 , an explanation on the depth frame numbers "going back" -
In general the frame number advance in sequential manner and should not revert.

In the above case, as you've configured both RGB and Depth streams to the same FPS an additional feature called **internal sync** kicks in. This feature causes RGB sensor to generate a HW pulse when capture starts to trigger Depth frame generation. Thus to ensure that the pair is captured simultaneously.  Note that in that pair RGB acts as a Master and Depth being the slave.
During the activation the Depth stream is reconfigured to switch into the Slave mode, and one of the attributes of this change is the frame counter being reset and starts back from zero.

So it is OK to see this behavior, actually in most cases it is even desirable as temporal synchronization is imperative for the most requires scenario of texture mapping/ alignment.

This feature has been brought up during several discussions here, e.g. #5867.  
Feel free to ask if you need it further elaborated