realsenseai/librealsense #6131: readme.md - Build with vcpkg instructions wrong for Windows

mediadog, 2020-03-25:
It says:

git clone https://github.com/Microsoft/vcpkg.git
cd vcpkg
./bootstrap-vcpkg.sh
./vcpkg integrate install
./vcpkg install realsense2

which is not correct for Windows; the /'s should be \'s, and the .sh should be .bat
Also, it should note that the 'vcpkg integrate install' should be executed as administrator.

MartyG-RealSense, 2020-03-25:
I have added the 'documentation' tag to your message so that RealSense team members responsible for docs can track it.  Thanks for the report!

MartyG-RealSense, 2022-05-16:
Hi @mediadog  As the documentation change request that you made in 2020 is now 2 years old, can you confim whether you still believe that the requested change should be made, please?  Thanks!

mediadog, 2022-05-23:
Well yes since simply looking at the readme shows the exact same instructions still there.  A better question is WHY HAVEN'T these changes been made after TWO YEARS?

MartyG-RealSense, 2022-05-23:
I am not involved in documentation changes, so I do not have information about that.  I will keep this documentation ticket open though.  Thanks very much for the update!

closed 2026-10-23 by MartyG-RealSense