realsenseai/librealsense #6132: Failed to launch camera

tispratik, 2020-03-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.12.03.00| 
| Operating System & Version |   Ubuntu 16.04 | 
| Kernel Version (Linux Only)    |  4.15.0-88-generic                                         | 
| Platform                                 |    - |
| SDK Version                            |  -                          | 
| Language                            |  python                         | 
| Segment			|  Robot                          | 

### Issue Description
The cameras in our system stopped working. We are getting the following error when trying to start the camera.

    for index, device in enumerate(devices):
RuntimeError: set_xu(ctrl=1) failed! Last Error: Resource temporarily unavailable

I tried hardware reset, that didn't help.

Hardware reset script:
```python
import pyrealsense2 as rs
config = rs.config()
pipeline = rs.pipeline()
profile = pipeline.start(config)
device = profile.get_device()
device.hardware_reset()
```

tispratik, 2020-03-25:
```
[Wed Mar 25 22:57:12 2020] usb 2-3: USB disconnect, device number 2
[Wed Mar 25 22:57:12 2020] usb 2-3: new SuperSpeed USB device number 4 using xhci_hcd
[Wed Mar 25 22:57:12 2020] usb 2-3: New USB device found, idVendor=8086, idProduct=0ad3
[Wed Mar 25 22:57:12 2020] usb 2-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Wed Mar 25 22:57:12 2020] usb 2-3: Product: Intel(R) RealSense(TM) Depth Camera 415
[Wed Mar 25 22:57:12 2020] usb 2-3: Manufacturer: Intel(R) RealSense(TM) Depth Camera 415
[Wed Mar 25 22:57:12 2020] usb 2-3: SerialNumber: 846623022810
[Wed Mar 25 22:57:12 2020] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[Wed Mar 25 22:57:12 2020] uvcvideo: Unable to create debugfs 2-4 directory.
[Wed Mar 25 22:57:12 2020] uvcvideo 2-3:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[Wed Mar 25 22:57:12 2020] uvcvideo 2-3:1.0: Entity type for entity Processing 2 was not initialized!
[Wed Mar 25 22:57:12 2020] uvcvideo 2-3:1.0: Entity type for entity Camera 1 was not initialized!
[Wed Mar 25 22:57:12 2020] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/input/input18
[Wed Mar 25 22:57:12 2020] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[Wed Mar 25 22:57:12 2020] uvcvideo: Unable to create debugfs 2-4 directory.
[Wed Mar 25 22:57:12 2020] uvcvideo 2-3:1.3: Entity type for entity Processing 7 was not initialized!
[Wed Mar 25 22:57:12 2020] uvcvideo 2-3:1.3: Entity type for entity Extension 8 was not initialized!
[Wed Mar 25 22:57:12 2020] uvcvideo 2-3:1.3: Entity type for entity Camera 6 was not initialized!
```

tispratik, 2020-03-25:
Output from Debug logs:

```
25/03 23:25:48,432 DEBUG [139642935887616] (context.cpp:108) Librealsense VERSION: 2.33.1
25/03 23:25:48,456 DEBUG [139642935887616] (ds5-factory.cpp:938) try_fetch_usb_device(...) failed.
25/03 23:25:48,456 DEBUG [139642935887616] (ds5-factory.cpp:938) try_fetch_usb_device(...) failed.
25/03 23:25:48,461 DEBUG [139642935887616] (ds5-factory.cpp:938) try_fetch_usb_device(...) failed.
25/03 23:25:48,461 DEBUG [139642935887616] (ds5-factory.cpp:938) try_fetch_usb_device(...) failed.
25/03 23:25:48,462 DEBUG [139642935887616] (rsusb-backend.cpp:34) Creating UVC Device from path: 2-5-3
25/03 23:25:48,572 WARNING [139642586076928] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
25/03 23:25:48,632 WARNING [139642586076928] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
25/03 23:25:48,692 WARNING [139642586076928] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
25/03 23:25:48,753 WARNING [139642586076928] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
25/03 23:25:48,813 WARNING [139642586076928] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
25/03 23:25:54,624 WARNING [139642935887616] (types.cpp:49) set_xu(ctrl=1) failed! Last Error: Resource temporarily unavailable
25/03 23:25:54,656 WARNING [139642935887616] (rs.cpp:283) null pointer passed for argument "device"
```

weblackey, 2020-03-26:
wondering if this is similar to the messages encountered in #6062 and #6112 ?

Same unresolved root cause of "Resource temporarily unavailable" ?

Did this happen for you after adding more USB devices or plugging into a different USB port ? Or after firmware or library version upgrade ?

tispratik, 2020-03-26:
It randomly happened. The only change we had deployed was to cater to the problem of "Frames didn't arrived" during `wait_for_frames()`. We had added an exception handling clause to initialize the camera... but forgot to close the earlier pipeline. Not sure if that had an adverse effect.

We tried many things to recover from this, a system reboot also didn't help. What helped was a system shutdown, with a 10 seconds wait and then bootup. Then the issue was fixed.

RealSenseCustomerSupport, 2020-03-31:
----------------------------------------------

@tispratik What platform and SDK version did you get the issue? Did you see this issue with RealSenseViewer or python example in SDK? Looking forward to your reply. Thanks!

tispratik, 2020-03-31:
SDK version `2.33.1`. Can you guide on how to check platform version? We're using Realsense D415.

RealSenseCustomerSupport, 2020-03-31:
----------------------------------------------

@tispratik Is your platform special one such as jetson nano or x86 PC?

tispratik, 2020-04-02:
No, ours is a regular GPU machine with NVIDIA 2080 and core i9 processor, 16 GiB RAM.

RealSenseCustomerSupport, 2020-04-08:
----------------------------------------------

@tispratik Did you see this issue with RealSenseViewer or python example in SDK? Or you only got issue with your own python script? Looking forward to your update. Thanks!

tispratik, 2020-04-08:
The issue occurred when i tried to enumerate devices using a python script. The issue is likely related to Ubuntu 16.04, i had not come across the issue on Windows with similar hardware.

RealSenseCustomerSupport, 2020-04-08:
----------------------------------------------

@tispratik So I understood the issue happened after the change you mentioned that "added an exception handling clause to initialize the camera... but forgot to close the earlier pipeline."  Could you please try if still have issue if close the earlier pipeline?   Also, what's the output result of "lsusb | grep 8086" when the issue happened?

RealSenseCustomerSupport, 2020-04-27:
----------------------------------------------

@tispratik Could you please update?Thanks!

tispratik, 2020-04-27:
The issue is not happening now, we did a reboot. But should be investigated or atleast a debug script should be provided so that we can easily provide all the required info while opening the issue.

RealSenseCustomerSupport, 2020-04-28:
----------------------------------------------

@tispratik  Thanks for the update!  Could you please share your python script or code snippet when get the issue?

RealSenseCustomerSupport, 2020-05-14:
----------------------------------------------

@tispratik Could you please update if you still need support on this?

RealSenseCustomerSupport, 2020-05-19:
----------------------------------------------

@tispratik Sorry we didn't get feedback from you for weeks. Will close this at this point. Please feel free to create another new one if you still have questions or issues. Thanks!

closed 2020-05-19 by RealSenseCustomerSupport