realsenseai/librealsense #6138: T265 pose drift

RendijsSmukulis, 2020-03-26:
-------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Operating System & Version |  Win 10 | 
| Platform                                 | PC  |
| SDK Version                           |  2.33.1 }                          | 
| Language                              |  unity                          | 

### Issue Description

I'm seeing a fair amount of constant pose drift on one of the axis of my T265. 

This is pose.translation data from a T265 sitting perfectly still:

[image]

The orange line represents the Y translation. It keeps constantly drifting at the same rate for ~6 measurements, and then "jumps" back, only to start drifting again on the next measurement, in the end giving the saw-tooth pattern. The data is recorded at 200fps, so I assume the jumps in data occur when the fisheye frame arrives and SLAM algo re-positions the pose data to fix the drift.

Here are the velocity measurements:

[image]

note how the X and Z axis average to approx 0 velocity, while Y keeps reporting negative velocity all the time. 

Is this to be expected? Is there a way to re-calibrate the IMU to reduce the constant drifting?

RendijsSmukulis, 2020-03-27:
This has currently been verified on two T265 devices, with the same results. The incorrect Y velocity also makes it impossible to do decent pose prediction.

RendijsSmukulis, 2020-03-27:
In fact, we've discovered some of our T265s are significantly better or worse than others. 

Here's a particularly bad case:
[image]

And here's a fairly good one:
[image]

Y-axis in meters, both graphs have the same scale.

neilyoung, 2020-03-27:
What is the scale on the left?

RendijsSmukulis, 2020-03-27:
> What is the scale on the left?

Meters. So the worst unit we have is jittering by about a millimeter every 30ms, while the best one keeps it under 0.2mm.

neilyoung, 2020-03-27:
I would think this is intrinsic to the algorithm. I guess you will be forced to filter that outside.

RendijsSmukulis, 2020-03-27:
I disagree. The devices are also constantly reporting incorrect acceleration and velocity values (0.03m/s velocity and 0.3m/s^2 acceleration from the worst device I posted above) when they are static.

neilyoung, 2020-03-27:
I believe you. I have seen this with my application too (https://neilyoung.serveblog.net). However, I have serious doubts, that anyone of the contributors/developers will do anything against this. I have opened so many issues, and just a handful of it have been resolved. I don't know. The T265 seems to be either a test balloon of Intel. Here in this forum it doesn't play any remarkable role, not many do really care. They are all focused on the depth cams. I find that sad, but it is as it is. So I suppose, you _have_ to live with it or put the cam aside. I know other SLAM solutions too and compared to them the T265 is gold. 

There is no hope for a fix or change soon.

masterchop, 2020-06-05:
so sad to spend so much money and find this out. I thought i was going to get some advises on how to reduce drift. I guess is good enough to play with.

mpottinger, 2020-09-28:
@neilyoung Apple's ARKit, Google ARCore and Huawei AREngine are the best SLAM solutions that I have seen. Especially Huawei AREngine is solid and Huawei phones have integrated depth sensors.

I have made realtime point cloud scanning work very easily on the Huawei P30 Pro. 

I wish there was an external depth camera that could do the same depth + tracking with that level of performance, especially since Huawei is in hot water right now, but none of this stuff comes close to what Huawei already has in their phones.

amr17, 2023-12-31:
> I believe you. I have seen this with my application too (https://neilyoung.serveblog.net). However, I have serious doubts, that anyone of the contributors/developers will do anything against this. I have opened so many issues, and just a handful of it have been resolved. I don't know. The T265 seems to be just a test balloon of Intel. Here in this forum it doesn't play any remarkable role, not many do really care. They are all focused on the depth cams. I find that sad, but it is as it is. So I suppose, you _have_ to live with it or put the cam aside. I know other SLAM solutions too and compared to them the T265 is gold.
> 
> There is no hope for a fix or change soon.

Hey,
I'm facing some issues with the T265. What alternatives would you recommend to replace it?

closed 2026-10-23 by MartyG-RealSense