realsenseai/librealsense #6141: Pyrealsense2 and wxPython bug

alaricLEA, 2020-03-27:
OS: Win 10 Home 1909
Python: 3.7.3
wxPython: 4.0.7 push2
Pyrealsense2: 2.33.1.1388

Hi. Does anybody know why importing pyrealsense2 causes wxPython to freeze the GUI and stop showing certain widgets? Sample code below. Suggestions most welcome.

import wx
import pyrealsense2 as rs   # Commenting this out will produce the dialog

class Main():

    def __init__(self):
        self.app = wx.App(False)       
        self.frame = wx.Frame(None, -1, '', size=(1000,500))
        self.frame.Centre()
        self.frame.Show(True)
        
        dlg = wx.DirDialog(None, 'If you see this then its working')
        dlg.ShowModal()

main = Main()

MartyG-RealSense, 2020-03-29:
There is very little information the subject of using wxPython with Pyrealsense2.  I did find one example though:

https://github.com/ADS-Astral/PaintBot/commit/fbdc6ecff35ab650da01b3169eba30d47a91e38b#diff-b60c9333256bbf4a67e0d2878e9a9400

In that script, the order of the import statements at the start of the script is reversed in comparison to yours, with **import pyrealsense2 as rs** being listed first:

**import pyrealsense2 as rs**
**import wx**

alaricLEA, 2020-03-29:
Thanks for the suggestion Marty. I'm afraid it didn't work.

MartyG-RealSense, 2020-03-29:
Have you successfully installed the Pyrealsense2 wrapper please before trying your script?

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python

alaricLEA, 2020-03-29:
Yes I've been driving my depth camera software with Python for a few months now and found this bug by chance.

I've also tried with tkinter (the standard Python GUI) rather than wxPython and this also causes the same issue.

Any ideas why Pyrealsense2 is interfering with Windows dialogs? Seems very odd.

MartyG-RealSense, 2020-03-29:
I am on the RealSense technical support team rather than the developer team, so it is a question best referred to the developers.  I will tag a couple of them into this discussion and also label this case as "Bug" to make sure that the developers can track it.  @dorodnic  @ev-mp 

I apologise for the delay in resolving your problem in the meantime.

alaricLEA, 2020-03-31:
Thanks for the assistance Marty. I hope there's a fix!

alaricLEA, 2020-04-27:
Solved in the other thread. #6174

closed 2020-04-27 by alaricLEA

MartyG-RealSense, 2020-04-27:
Awesome news - thanks for the update!