realsenseai/librealsense #6142: Function create_pointcloud not found

ViacheslavRud, 2020-03-27:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | n/a | 
| Firmware Version                   | n/a | 
| Operating System & Version |   Linux (Ubuntu 16.04)  | 
| Kernel Version (Linux Only)    |  4.15.0-45-generic                                         | 
| Platform                                 | n/a  |
| SDK Version                            |  2.33.1                         | 
| Language                            | C++ 11                          | 
| Segment			|   (does not matter, fails in simple standalone app)                          | 

### Issue Description
Hi. I'm trying to use this code
```
#include <librealsense2/rs.hpp>
...
rs2::pointcloud pc = rs2::context().create_pointcloud();
```
from this tutorial [https://github.com/IntelRealSense/librealsense/tree/master/examples/pointcloud](https://github.com/IntelRealSense/librealsense/tree/master/examples/pointcloud).
But it gives me next error:
`class rs2::context’ has no member named ‘create_pointcloud’`

What is wrong here?

MartyG-RealSense, 2020-03-28:
This sample is designed to call upon OpenGL graphics, so it may not be the best example to adapt if you are trying to reduce the program's complexity down to a simpler point cloud application.  For example, your script omits the **#include "example.hpp"** line, and example.hpp contains instructions for rendering a point cloud with OpenGL.

The simplest form of a C++ point cloud script, which may be a better place to begin experimentation at, is this:

`rs2::pointcloud pc;
rs2::points points = pc.calculate(depth_frame);
pc.map_to(color_frame);`

A slightly more complicated version of this that can export the point cloud to a .ply format file is in the link below.

https://github.com/IntelRealSense/librealsense/issues/2488#issuecomment-427814215

ViacheslavRud, 2020-03-28:
> This sample is designed to call upon OpenGL graphics, so it may not be the best example to adapt if you are trying to reduce the program's complexity down to a simpler point cloud application. For example, your script omits the **#include "example.hpp"** line, and example.hpp contains instructions for rendering a point cloud with OpenGL.
> 
> The simplest form of a C++ point cloud script, which may be a better place to begin experimentation at, is this:
> 
> rs2::pointcloud pc;
> rs2::points points = pc.calculate(depth_frame);
> pc.map_to(color_frame);
> 
> A slightly more complicated version of this that can export the point cloud to a .ply format file is in the link below.
> 
> [#2488 (comment)](https://github.com/IntelRealSense/librealsense/issues/2488#issuecomment-427814215)

Thank you. But I have additional question then. I will describe in more details.
I need to take depth data from .bag file and convert it to point cloud. It should work under ROS.
It should emulate real camera which produce pointcloud messages in topic **camera/depth_registered/points**. But .bag record does not contain this data. So I have to convert depth to point cloud pcl::PointCloud<pcl::PointXYZRGB> type.
I found this tutorial in thread https://github.com/IntelRealSense/librealsense/issues/2090

Seems it use **sensor_msgs/Image** message in **devise_0/sensor_0/Depth_0/image/data** topic. And your code use **frame** type.
How to convert message to frame?

P.S. I dont't use rendering part from tutorial, I need only conversion.

MartyG-RealSense, 2020-03-28:
If you need to convert ROS data to PCL data, this link may be helpful:

https://answers.ros.org/question/316701/ros-intel-realsense-how-to-process-data-with-pcl/

Alternatively, if you are able to make use of the Python language then there is an Intel tutorial for loading in a bag file, retrieving the data from it and generating a point cloud.

https://github.com/dorodnic/binder_test/blob/master/pointcloud.ipynb

dorodnic, 2020-03-29:
There's an error in documentation of pointcloud, I'll update it for the next release.

ViacheslavRud, 2020-03-29:
**dorodnic**, ok, thanks.

Thanks for links, **MartyG-RealSense**,  I will try to use them.

MartyG-RealSense, 2020-06-18:
Hi @ViacheslavRud   I reviewed the pointcloud documentation, and an edit for **create_pointcloud()** was made on March 29 2020.  

https://github.com/IntelRealSense/librealsense/commit/45f86e523d224b8b995e42a78bd3619fa7bdb7b2

I will therefore close this case as completed.  Thanks!

closed 2020-06-18 by MartyG-RealSense